prusa3d/PrusaSlicer #5314: 2.3 beta 1 - Export failed

paulamath, 2020-11-28:
### Version
PrusaSlicer 2.3.0-beta1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Stella 2

### Behavior
After slicing the .stl and exporting to the SD card, the message "failed to export" appears. The same is true if you try to export to any folder on your computer.

The same was true in the Alpha 4 version.

#### Project File (.3MF) where problem occurs
https://we.tl/t-CdpB3ya6TN
[image]

[benchyError.zip](https://github.com/prusa3d/PrusaSlicer/files/5611082/benchyError.zip)

kocikdav, 2020-11-30:
Hello, thank you for reporting. This seems to be a problem with translation, the original message is "Exporting finished". Would you please confirm that the file is on your SD card?

klwill1192, 2020-11-30:
I'm experiencing a similar problem.  When I attempt to export GCODE to an SD card, PrusaSlicer tells me the SD card is write protected.  It is not--I can save GCODE from Cura to the SD card.  In fact, I can save from Cura, then immediately attempt to save from PrusaSlicer (failing), then go back and save again from Cura.  I can also drag and drop files from other sources onto the card.  PrusaSlicer seems to be the only application having a problem writing to the card.

kocikdav, 2020-12-01:
> I'm experiencing a similar problem...

@klwill1192  Hello, that is not a similar problem. This issue is about wrong translation and exporting is working fine. If you are having problem exporting, please fill new issue. Best with exact error message.

closed 2020-12-01 by kocikdav

paulamath, 2020-12-02:
Using the same thing in English!

bubnikv, 2020-12-02:
> Using the English version, the same error occurs! Yes, the file is saved correctly!

The Windows error messages are reported by boost::filesystem using the system language, which did not change by you switching the language in PrusaSlicer.

I am aware that it is a difficult topic, we have difficulties to grasp it ourselves, and it behaves differently on each platform (Windows, OSX, Linux).

kocikdav, 2020-12-02:
@paulamath If by same error you mean Exporting finished notification (as on picture) than it is not error but notification about successful export with possibility to eject your removable media. And everything is alright :). If there is a error, there would be a popup dialog with stated reasons of failure. 

[image]

bubnikv, 2020-12-05:
Guys, please retest with the upcoming PrusaSlicer 2.3.0-beta2. Thank you.

@klwill1192
What operating system?
Would you also retest with PrusaSlicer 2.3.0-beta2? Thanks.