prusa3d/PrusaSlicer #5318: 2.3.0 Beta 1 Feature Request Firmware Settings

GrubbyZebra, 2020-11-29:
### Version
2.3.0 Beta 1

### Operating system type + version
Windows x64 Pro

### Behavior
Feature Request: With the advent of the new "physical printer" settings, suggest moving the "firmware" settings to the "physical printer" instead the "printer settings" tab, since this information will not change with nozzle size changes.

rtyr, 2020-11-30:
From Alpha1 release notes (https://github.com/prusa3d/PrusaSlicer/releases)

> Physical Printers settings
> 
> Historically PrusaSlicer stored the printer network connection settings into the Printer profile. This became inconvenient with the introduction of read-only vendor specific system profiles, where the user had to derive his own Printer profile from a system one just to be able to configure G-code upload over the network. In addition, storing the network connection information into the Printer profile posed a security risk, which we had to patch by suppressing export of the network connection into the 3MF, G-code and Config Bundles anyway.
> 
> Starting with this version, PrusaSlicer separates the physical printer connection settings from the Printer profile into new Physical Printer profiles. **A Physical Printer profile references one or more Printer profiles of the same technology (either FFF or SLA), and it assigns a network connection information to these printers.** One does not have to create a Physical Printer profile if one does not need to connect to a printer over the network. The new Physical Printers are listed in a new section in the Printer selection combo box after the system and custom Printer profiles.
> 
> A new Physical Printer profile is created by clicking on the "cog" icon right to the Printer profiles combo box, or by selecting the "add or remove printers" item in the Printer combo box. The Physical Printer profile editor opens also when clicking on the new "cog" icon in the Printer settings tab. The Physical Printer profiles are being stored into PrusaSlicer/physical_printer directory.
> 
> When first installing PrusaSlicer 2.3.0-alpha1, the network connection information is removed from existing user Printer profiles and converted to new Physical Printers profiles giving them generic names "Printer 1, Printer 2, ...". The user is advised to review and rename these Physical Printer profiles generated by the conversion process. The following configuration keys are moved from Printer to Physical Printer profiles: "host_type", "print_host", "printhost_apikey", "printhost_cafile", "printhost_authorization_type", "printhost_user", "printhost_password". The last three keys are new to support SL1 HTTP digest authentication, see the following section.
> 
> If some parameter of a Printer profile is modified, a "(modified)" suffix is newly shown not only for the modified Printer profile in the list of Printer profiles, but also for all the associated Physical Printer profiles.
> 
> For security reasons, the "Export Config Bundle" function does not export the Physical Printers. To help sharing the complete PrusaSlicer profiles between ones computers, we added "Export Config Bundle with Physical Printers" option.

bubnikv, 2020-12-07:
Thanks but no.

closed 2020-12-07 by bubnikv