prusa3d/PrusaSlicer #5319: If you select import configuration then it will blow away all the default configurations.

mhfleming, 2020-11-29:
### Version
Version 2.3.0-beta1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 Version	10.0.19042 Build 19042

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3S, latest firmware

### Behavior
* _Describe the problem_
If you select import configuration then it will blow away all the default configurations.
* _Steps needed to reproduce the problem_
1) In a previous version, export you configuration bundle
2) Open new version
3) Go through Configuration Wizard
4) Check to make sure you see all the expected print, filament and printer settings
5) Select Import->Import Config Bundle
6) Select file from step 1
7) You should see a dialog telling you how many presets were imported, in my case 13
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
I expected to see all the default configurations along with my imported ones
* _Actual Results_
I only see some of my imported ones.
[PrusaSlicer_config_bundle.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/5611496/PrusaSlicer_config_bundle.ini.zip)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
After the configuration wizard
[image]
After the import has completed
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-12-08:
I am not sure what is the issue here. It may be that we have renamed the MK3S printer profile to MK3S & MK3S+, while we have forgotten to add the old name to the list of renamed_from in the renamed profile.
@rtyr  will look into it.

bubnikv, 2020-12-08:
Confirmed with @rtyr , the renamed_from fields were filled in correctly.
If you switch to MK3S+ printer, will you see your profiles?

bubnikv, 2020-12-08:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/6e3fae386533843baadbf03fcf18ec294ebcb968
This issue was there since the introduction of system profile. The issue was caused by us renaming the system profiles and not testing for the old profile names in the "inherits" fields of the profiles loaded from your config bundle.

closed 2020-12-08 by bubnikv

mhfleming, 2020-12-08:
It is so cool that my feedback helped!