prusa3d/PrusaSlicer #5320: [Bug] MMU2S Wipe Tower collides with Objects after Auto Arrange

Area5142, 2020-11-29:
### Version
PS 2.3.0 Beta1

### Operating system type + version
Linux Mint 19.3 x64

### 3D printer brand / version + firmware version (if known)
Prusa MMU2S

### Behavior
The Wipe Tower collides with the objects after auto arrange.

1. Begin MMU2S empty project.
2. Add two standard cubes and assign different color to them.
3. Multiply cube count by 3 to get 6 cubes total.
4. Set Auto Arrange Gap Size to 1.317 mm (It is very difficult to set precise size with slider)
5. Auto Arrange objects.

Gap Size:
[image]

Layout after auto arrange (colliding):
[image]

Preview (colliding):
[image]

_Expected behavior:_
Gap of at least 1.327 mm from Wipe Tower to objects.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Wipe Tower Collide - 2.3.0 Beta 1.zip](https://github.com/prusa3d/PrusaSlicer/files/5611499/Wipe.Tower.Collide.-.2.3.0.Beta.1.zip)

closed 2020-12-07 by tamasmeszaros

Area5142, 2020-12-08:
Why close this issue?
Just tested with 2.3.0 Beta 2 and it is still a problem. Wipe Tower collides with objects on build plate when arrange offset is set to 1.5 mm or less:

[image]

There seems to be room enough for the Wipe Tower, but it is placed wrong. Properly due to a calculation error.

bubnikv, 2020-12-08:
598ac290a1c4d305450252de7422c224966bb494 is after beta2

commit 598ac290 (2020-12-07): Fix for bad wipe tower handling in arrangement

fixes #5320