prusa3d/PrusaSlicer #5324: Feature Request Printer name in output

dtest505, 2020-11-29:
### Version
2.2.0+

### Operating system type + version
Mac OSX 10.13

### 3D printer brand / version + firmware version (if known)
Multiple with mariln

### Behavior
Would be nice to have a setting for printer name and also include that in output options.
Having different printers it would make it easier to tell what .gcode file gets sent to which printer.

rtyr, 2020-11-29:
It is already possible.

[image]

https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)

dtest505, 2020-11-29:
*almost*.  Prefer just a printer name, not the full printer settings ID.  I just need to see its for the, say CR10, but that printer has 3 different settings (generic, PETG, TPU)

Guess I'll just do more bash scripting to auto fix it.

bubnikv, 2020-11-30:
[printer_model]

dtest505, 2020-12-02:
If its the model in Printer Settings > General > Bed shape > Set > load model
That doesn't work

closed 2020-12-02 by dtest505