prusa3d/PrusaSlicer #5328: PS does not respond to speed changes via 3DConnexion driver with supported device

tg73, 2020-11-29:
### Version
2.3.0-beta1 Win x64

### Operating system type + version
Win10 x64

### Behavior
According to the release notes for beta1:

> For the devices supported by the 3DConnexion driver it is recommended to configure the 3DConnexion settings in the 3DConnexion driver.

I'm using a [SpaceMouse Pro](https://3dconnexion.com/uk/product/spacemouse-pro/), which is a current and supported device. Changing the speed/scaling settings in the 3DC settings app has no effect in PS. Even with all the possible speed sliders to max in the 3DC settings app, PS is still as slow to move as by default. Legacy support for 3DC is unchecked in PS prefs.

[image]

[image]

[image]

Enabling legacy 3DC support, and using CTRL+M menu *does* work. However, for some reason the device name is prefixed with LOGITECH:

[image]

dahuafschmied, 2020-11-30:
Logitech is the manufacturer of the 3DConnexion devices. They changed the Vendor ID on newer devices to their own.

bubnikv, 2020-11-30:
@dahuafschmied 
> Logitech is the manufacturer of the 3DConnexion devices. They changed the Vendor ID on newer devices to their own.

Maybe @tg73 should update his drivers?

tg73, 2020-11-30:
I did some research. 3DConnexion has been a subsidiary of Logitech for many years (possibly from its inception). So yes, Logitech manufactures the hardware, but the drivers and R&D is done by 3DConnexion. As per one of the screenshots in the original post, my drivers are fully up to date, and this is confirmed by the current latest version on the 3DConnexion website.

foreachthing, 2020-11-30:
@tg73 my device even has a sticker on the bottom, saying it's a Logitech Company:
[image]

tg73, 2020-11-30:
@foreachthing Yours is fancy! Mine doesn't mention Logitech:
[image]

bubnikv, 2020-11-30:
> I'm using a SpaceMouse Pro, which is a current and supported device. 

Frankly I would not wonder if your device was no more supported by their current driver.

tg73, 2020-11-30:
It's a current product listed and actively promoted for sale on their [website](https://3dconnexion.com/uk/product/spacemouse-pro/). Here's a screenshot of the latest driver download showing the supported devices (my device annotated with red underline):
[image]

foreachthing, 2020-11-30:
> @foreachthing Yours is fancy! Mine doesn't mention Logitech:

No, just very old and very not-supported anymore :-[

enricoturri1966, 2020-12-01:
One way to know if the installed driver supports your device is to check the name of the application into the 3DConnexion settings dialog.
If it is set to "PrusaSlicer" the driver supports your device and automatically connects it to PrusaSlicer. (starting from version 10.7.0 of the driver)
If it is set to "prusa-slicer" there may be two cases: 
1) the devices is not supported (starting from version 10.7.0)
2) the driver does not associate automatically the device to PrusaSlicer (versions prior of 10.7.0)
This is an example: 
[image]

dahuafschmied, 2020-12-01:
Just tried my SpaceNavigator (a Logitech device) on Windows. Speed settings are working at the 3DConnexion driver. And PS is recognized an "PrusaSlicer". Version 10.7. So it is not a Logitech brand problem.

By the way: what about the Mac Support?? @enricoturri1966 #5312

enricoturri1966, 2020-12-01:
> By the way: what about the Mac Support?? @enricoturri1966 #5312 
We are working on it.

tg73, 2020-12-01:
@enricoturri1966 As per screenshot in the very first comment, I see "prusa-slicer". However, other apps work fine and respond correctly to speed changes in the 3DConnexion settings - for example, it works perfectly with Fusion 360. So this would appear to be a specific interop issue with PS. Are there any diagnostics I can enable is PS?
[image]

tg73, 2020-12-01:
@enricoturri1966 I am an experienced Windows developer (C++ many years ago, now mostly C#/.NET). I have got PS building locally in the past, so I can attempt some debugging if you can give me suggestion of where to breakpoint etc. In the mean time, this might be of interest. Spy++ shows PS plater window receiving messages like this when moving the SpaceMouse:

[image]

Fusion 360 main view receives messages like this:

[image]

bubnikv, 2020-12-01:
The 3DConnexion driver recognizes PrusaSlicer by its prusa-slicer.exe
executable name, and there needs to be an appropriate xml file installed as
part of the 3DConnexion driver. The latest 3DConnexion driver should
install such xml.
The XML file only registers for prusa-slicer.exe, not for
prusa-slicer-console.exe or prusa-gcodeviewer.exe

tg73, 2020-12-01:
@bubnikv Where should it put this XML file? Where is it on your test system?

tg73, 2020-12-01:
I have `C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg` as follows. Does it look correct?

```xml
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!--
(c) Copyright 3Dconnexion. All rights reserved. 
$Revision: 16358 $
-->
<AppCfg  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" Default="true" CfgFormatVersion="1.3" ThisFileVersion="1.0">
  <CfgProperties>
    <ID>ID_PrusaSlicer</ID>
    <Name>PrusaSlicer</Name>
    <InheritsFromID>ID_Default_KMJ_Cfg</InheritsFromID>
  </CfgProperties>
  <AppInfo>
    <Signature>
      <Name>PrusaSlicer</Name>
      <ExecutableName>prusa-slicer.exe</ExecutableName>
      <Transport>RawInput</Transport>
    </Signature>
  </AppInfo>
  <AuthorInfo>
    <DriverVersion>17.6.3.16588</DriverVersion>
    <Created>
      <Tool>3DxService.exe</Tool>
      <Version>17.6.3.16588</Version>
      <Date>2020:4:3:1:10:16:46:953</Date>
      <User>prusa</User>
    </Created>
    <LastChange>
      <Tool>3DxSmartUI.exe</Tool>
      <Version>1.2.19261.16572</Version>
      <Date>4/1/2020 10:22:47 AM UTC</Date>
      <User>prusa</User>
    </LastChange>
  </AuthorInfo>
</AppCfg>
```

tg73, 2020-12-01:
FYI, I uninstalled 3DxWare 10.7.0, restart PC, the file `C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-slicer.xml` is gone, reinstalled from `3DxWare64_v10-7-0_r3248.exe` (SHA1 `E76E60864665D0CCF0B1DF1588A566EF04AA8A9B`), `prusa-slicer.xml` is back, exactly the same as before the uninstall, PS behaviour unchanged.

bubnikv, 2020-12-01:
Tom, frankly I don't know. Maybe you shall try your luck with their support?

tg73, 2020-12-01:
tl;dr: `3DxService.exe` does not like executable names containing `-` characters, at least under certain undetermined circumstances.

Ok, after lots of hacking with SysInternals tools, dnSpy etc etc, it seems that:

- `3DxService.exe` handles app focus changes, finds the matching app config based on executable name, and updates the file `%LOCALAPPDATA%\3Dconnexion\3DxWare\3DxServiceState.xml` on disk. Various other 3Dx process watch `3DxServiceState.xml` for updates, and act accordingly, including `3DxSmartUi.exe`, which is the settings app.
- `3DxService.exe` looks for an exact process name match from all discovered configs in `C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\*.xml`, and if no exact match is found, it falls through to a regex match as defined in `AppDefCfg_KMJ.xml`, which has `/AppCfg/AppInfo/Signature/ExecutableName` = `^(.*)$`.
- On my PC, when PS gets focus, `3DxService.exe` behaves as though it does not find an exact process name match, and falls through to matching `AppDefCfg_KMJ.xml`.
- I tried a great many things to try to get the match to work. I noticed that 'prusa-slicer.exe' was the only exe name in all the config files containing a `-` character. At the time of writing, the only thing that works is to rename `prusa-slicer.exe` to something without a `-` (eg, `prusa_slicer.exe`), and change `prusa-slicer.xml:/AppCfg/AppInfo/Signature/ExecutableName` also to `prusa_slicer.exe`. Then it works. Even though `-` is not a reserved character in regex, I have tried various more or less permissive regexes for the exe name in `prusa_slicer.xml`, but none match and my assumption is that `3DxService.exe` does not treat it as a regex - perhaps it treats `^(.*)$` as a special case string.

This seems like a 3Dx bug to me. Assuming that you have a developer relationship with 3Dx, perhaps you could raise this with them. I don't know why this seems to happen only to me. I suspect that it's not literally just my PC, but that the issue hasn't been noticed by others yet. My Windows install is pretty vanilla - I don't heavily tweak it. I'm running Win 10 Pro 1909 (18363.1198).

Might I also point out that `prusa-slicer.xml` is the only config file to have `<InheritsFromID>ID_Default_KMJ_Cfg</InheritsFromID>`. All the others have `<InheritsFromID>ID_Base_Cfg</InheritsFromID>`. This looks suspicious to me.

Also, renaming `prusa-slicer.exe` is not a viable workaround, because PS raises the following error when running up:

[image]

bubnikv, 2020-12-02:
@tg73 Hats off, Tom. You did a great detective work.

> dnSpy 

The tool is new to me. Does it mean that all the 3DConnexion services are written in C#?

> This seems like a 3Dx bug to me. Assuming that you have a developer relationship with 3Dx, perhaps you could raise this with them.

I will talk to them. We also need to add the prusa-gcodeviewer registration into their driver.

> I don't know why this seems to happen only to me. I suspect that it's not literally just my PC, but that the issue hasn't been noticed by others yet.

The dash may have a different representation in Unicode. Also I have quite a record of fighting with local code page / UTF8 / 16 bit encoding conversion issues. Would you please check that the dash is a real dash in all the important spots, and not some special Unicode "long dash" or whatever alternative it may be?

tg73, 2020-12-02:
@bubnikv Thanks!
>The tool is new to me. Does it mean that all the 3DConnexion services are written in C#?

`3DxSmartUi.exe`, the settings app, is a .NET assembly, so most likely originated using C#. That helped me to identify the means of inter-process communication, and that the config lookup was done by whoever writes `3DxServiceState.xml`. All the other `3Dx*.exe` that I looked at were C++.

>The dash may have a different representation in Unicode.

This was one of the first things I looked at. I've been bitten by similar issues in the past - and worse, non-printing Unicode sequences. You only have to waste a day going mad tracking down such a problem once, and then you are careful to rule that out quickly in the future. I also tried using the `&#` entity form for the dash in `prusa-slicer.xml` in case it was some really odd XML parsing issue, but it didn't help.

bubnikv, 2020-12-12:
no response from 3DConnexion after 10 days.

Bruno969, 2020-12-27:
I have a SpaceMouse Wireless for 3DConnexion. It worked perfectly with PrusaSlicer 2.2.
With 2.3.0.RC2 I have the same issue: The speed change is not recognized. The View move very slow.

foreachthing, 2020-12-27:
@Bruno969 try it with ctrl+m, or legacy device settings in properties.

tg73, 2020-12-28:
Update: the problem persists with the latest 3DConnexion drivers 3DxWare 10.7.1.3280

Bruno969, 2020-12-28:
> @Bruno969 try it with ctrl+m, or legacy device settings in properties.

With "Enable support for legacy 3DConnextion devices" I got the ctrl+m Menu and with this i was able to change the speed of my SpaceMouse.
The SpaceMouse is not a legacy product, but this setting is needed to use it. Issue solved! 👍 
Many thanks!

bubnikv, 2021-01-06:
@tg73 No response from 3DConnexion until now. I am going to award them with "The worst business partner ever" badge.

bringbackgroupb, 2021-02-28:
@bubnikv any luck with 3DConnexion yet? I tried messing around with renaming the exe and modifying the config files, but I've had no luck getting it to work. I was really excited for the 2.3.0 release because I was finally going to have my SpaceMouse axes configurable, only it turns out there's a bug w/ 3DConnexion.

bringbackgroupb, 2021-03-17:
Just an update for anyone following this topic, it appears as of 3DxWare 10.7.1.3280 and PrusaSlicer-2.3.0+win64-202101111315, this issue has been resolved. I successfully inverted the axes on my SpaceMouse through the the 3DxWare driver and was also able to adjust speed. Big thanks to @bubnikv for getting this feature working!

tg73, 2021-03-23:
@bringbackgroupb It's still not working for me, with PS 2.3.0 release and 3DxWare 10.7.1.3280. In the screenshot below you'll see that the 3DxWare window still shows "prusa-slicer", not "PrusaSlicer" as it should, and indeed changing, for example, the speed in 3DxWare has no effect in PS. Do you see "prusa-slicer" or "PrusaSlicer"? Are you sure you don't have remnants of your efforts at exe renaming lying around?

[image]

bringbackgroupb, 2021-03-24:
@tg73 I have a brand new clean install of Window 10 Pro x64, so maybe that's what did it. My exe is still named "prusa-slicer.exe", but now 3DxWare is detecting it with the name "PrusaSlicer". I'm guessing they've updated their default profiles. You may need to fully remove all traces of the driver from you computer and reinstall to get it working.

[image]

bubnikv, 2021-03-24:
> @tg73 <https://github.com/tg73> I have a brand new clean install of
Window 10 Pro x64, so maybe that's what did it. My exe is still named
"prusa-slicer.exe", but now 3DxWare is detecting it with the name
"PrusaSlicer". I'm guessing they've updated their default profiles. You may
need to fully remove all traces of the driver from you computer and
reinstall to get it working.

You may need to remove the xml files you copied manually into the
3DConnexion driver directory before you upgrade the 3DConnexion driver. Or
just uninstall the 3DConnexion driver, delete all traces of the 3DConnexion
driver (including those XML files you copied there) and then install the
3DConnexion driver.

tg73, 2021-03-25:
@bubnikv So - there was a newer version of 3DxWare on their website which was not picked up by auto update. I uninstalled, completely removed `C:\Program Files (x86)\3Dconnexion`, rebooted, installed the latest version, and I still get the same behaviour (ie, you see `prusa-slicer`, not `PrusaSlicer`). I also checked for any 3Dx configuration or persistent state that might be stashed in the registry, but found none.

tg73, 2021-03-25:
I found some further config files from my diagnostic hacking in last year in `C:\Users\<username>\AppData\Roaming\3Dconnexion\3DxWare\Cfg`. I uninstalled 3DxWare, removed any prusa-like files from that directory, and reinstalled (with any requested restarts), and now its working! So I concur that the issue does appear to finally be fixed. Closing.

closed 2021-03-25 by tg73