prusa3d/PrusaSlicer #5332: Feature request: Allow per-filament overides for {filament_type[0]} output variable

evenprimes, 2020-11-29:
### Version
2.3+ (Feature request for a future version)

### Operating system type + version
All supported OS.

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Allow a per-filament override for what is output in the `{filament_type[0]}` replacement variable.

#### Current behavior
The `{filament_type[0]}` variable only outputs the basic type of filament such as "PLA" or "PETG".

#### Requested behavior
Create a new option (on the "Filament Overrides" tab seems the most obvious) where we can enter custom text for a given filament profile. This would allow easier gcode storage for fine tuned filament profiles.

The expected use case is for holding test models. It would be good to have the "Prusament PLA" gcode and the "3D Fuel" gcode both available and easy to tell them apart.

Some sample filenames"
* `benchy_0.2mm_Prusament PLA_MK3S_1h21m.gcode`
* `benchy_0.2mm_3dfuel PLA_MK3S_1h21m.gcode`
* `benchy_0.2mm_protopasta PLA_MK3S_1h21m.gcode`
* `benchy_0.2mm_filamentone matte PLA_MK3S_1h21m.gcode`

rtyr, 2020-11-29:
> Some sample filenames"
> 
>     * `benchy_0.2mm_Prusament PLA_MK3S_1h21m.gcode`
>     * `benchy_0.2mm_3dfuel PLA_MK3S_1h21m.gcode`
>     * `benchy_0.2mm_protopasta PLA_MK3S_1h21m.gcode`
>     * `benchy_0.2mm_filamentone matte PLA_MK3S_1h21m.gcode`

Use `filament_settings_id[0]` if you want filenames like this.

You can also use your own custom filament types.
[image]

evenprimes, 2020-11-29:
I don't remember seeing `filament_settings_id[0]` when I looked in the docs before and I can't find the page now, so thanks, this is exactly what I wanted.

It never even occurred to me to try typing in a different filament type, so that's pretty cool, too!

closed 2020-11-29 by evenprimes