prusa3d/PrusaSlicer #5333: Possibility use cursor keys to move in sliced object

ochm, 2020-11-29:
### Version
2.3.0-beta1+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
In version 2.2.0 there was possibility to move in sliced object by layers up&down using cursor keys up and down.

In 2.3.0-beta1 is this possible only when the mouse is over vertical ruler, and in the horizontal ruler there is no possibility use cursor keys left and right.

Is it possible to return back the possibility use cursor keys up&dow and newly possibility to use left&right for horizontal ruler, without mouse over any ruler?

_Is this a new feature request?_
Maybe

#### Project File (.3MF) where problem occurs
Not needed.

fe60, 2020-11-30:
The issue regarding the up&down navigation has already been reported in #5192 
To not mix up bug reports and feature requests, I recommend to create a separate issue requesting the left&right navigation.

bubnikv, 2020-12-07:
This should work in PrusaSlicer 2.3.0-beta2, see the "keyboard shortcuts" dialog when pressing the "?" hotkey in 3D scene.

closed 2020-12-07 by bubnikv