prusa3d/PrusaSlicer #5337: [2.3.0 Beta-1] Misleading dialogue box in configuration wizzard

Boby71, 2020-11-29:
### Version
2.3.0-Beta1

### Operating system type + version
Win10/64Bit

### 3D printer brand / version + firmware version (if known)
MK2.5S

### Behavior
This dialogue here...
[image]
...is somewhat strange and misleading. 

Because if I click "No", then I expect the wizzard to continue WITHOUT installing standard filaments (e.g. because I want to import an existing config lager). But when I click "No", nothing happens - I stay on this screen.
When I click "Yes", then a default selection will be applied and I am able to continue.

So either my expectation is wrong - or this dialogue is useless, because if it's not possible to choose "No", then better present the user a dialogue box with the information "default selection has been applied" with a simple OK button.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Not necessary

rtyr, 2020-11-30:
> Because if I click "No", then I expect the wizzard to continue WITHOUT installing standard filaments (e.g. because I want to import an existing config later). But when I click "No", nothing happens - I keep staying on this screen.
> When I click "Yes", then a default selection will be applied and I am able to continue.
> 
> So either my expectation is wrong - or this dialogue is useless, because if it's not possible to choose "No", then better present the user a dialogue box with the information "default selection has been applied" with a simple OK button.
> 
You need to enable at least one filament profile to continue, so it is correct behavior. You can either click "No" and select something or select "Yes" to enable basic set of filament profiles.

Boby71, 2020-11-30:
Ah, okay - I wasn't expecting that at least one has to be selected. I installed the beta and wanted to import the alpha settings, so I was a kind of puzzled...

closed 2020-11-30 by Boby71