prusa3d/PrusaSlicer #5338: [Feature Request]

gutenbar, 2020-11-29:
### Version
2.2.0
### Operating system type + version
Windows 10, 64 bit, b. 19041.630
### 3D printer brand / version + firmware version (if known)
Original Prusa MK3s, firm. 3.9.0.3421

### Behavior
Ok. We have the "Don't support bridges" option. But how to tell the Slicer how far it should be considered as a bridge?

I designed a structure considering the bridge concept. So, almost all faces have high angles with some bridges. The exceptions are some faces on the top, expected to be supported. Although its behavior is correct with very small bridges (it did not support them), **PrusaSlicer insists in want to also support those not so bigger bridges (1 mm or 2 mm length)**. (See attached figure).

**I think it could be a good idea an option informing the bridge maximum length without supporting it.**

[image]

bubnikv, 2021-02-19:
Please provide a 3MF.

bubnikv, 2021-10-05:
No response, also there is an issue asking for maximum unsupported bridge length.

closed 2021-10-05 by bubnikv