prusa3d/PrusaSlicer #5339: Arrange with rotation does not work

Fliens, 2020-11-29:
### Version
2.3.0 Beta

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Prusa MINI

### Behavior
When right clicking on the "arrange" option in the slicer and activating the "rotate" option nothing is arranged or rotated!
I've imported two models that would fit on the printbed if rotated but the slicer does nothing.
I only get a message that an object outside of the printing area was detected and that i have to resolve the problem to continue slicing(bottom right corner of the image)
[image]

As you can see in the next picture, it is possible to fit them on the printbed(It looks like they are touching the white line but that's not the case -> they would be rendered blue):
[image]

The models are from the popular beer cap gun:
https://www.thingiverse.com/thing:3915647

tamasmeszaros, 2020-11-30:
Hi @Fliens! Arrangement doesn't take into account every possible rotation, that would take massive amount of processing time.

foreachthing, 2020-11-30:
@Fliens Even with two parts, at every single degree, you'd (or PS would) have to try a max. of 720 (2x360°) possibilities. Imagine four or forty parts.
However, this https://github.com/Jack000/SVGnest sounds very interesting! If PS could do that ........ 😎

tamasmeszaros, 2020-11-30:
SVGnest does a deep search and possibly repeats the arrangement multiple times as part of an optimization algorithm. PS does it in one shot which is very fast and produces a usable solution. I can only imagine the complaints when users have to wait minutes for the result.

Fliens, 2020-11-30:
I would wait a minute for a usable arrangement.
I have the MINI and because of the small print bed a method to find an optimized layout for the max amount of parts would be amazing!
Arranging those big parts that just fit (like on the picture) takes more than two minutes. And if the other method even improves the distance between the parts, that would take so much work of my shoulders. Because keeping an eye on the distance between the parts is really tricky.

Just having the option in the slicer would be great, even if the algorithm takes some time :)

tamasmeszaros, 2020-11-30:
Yes, it is under discussion. Certainly possible to let it do it's thing for a longer period, fully armed. It's very hard to estimate the needed time in that case. It might be better to show partial results and let the user stop it when he is satisfied with the arrangement.

ulab, 2020-12-10:
It would be great if it'd work for a single (limited number of) part though. 

I just stumbled over this, because I have a single big part I know fits on the bed diagonally, but rotating and moving it around manually until it fits probably takes a lot longer than any algorithm ;-)

ulope, 2022-04-13:
I agree with @Fliens and @ulab. It's my time and processing power to "waste". 

3D printing already trains us to wait for things and I'd much rather have my computer spend 2, 3 or 5 minutes arranging objects while I get a coffee than me manually fiddling around with the objects the same amount of time.

Now I totally agree that this should be a power user feature and there should be a way to cancel if it takes too long.
The idea of showing intermediate results and continuing sounds quite nice.

GhostDog98, 2023-10-29:
Is there any updates on this? I'm stuck with needing to use the svgnest tool after exporting my stl's to an svg...

closed 2026-09-29 by prusaslicer-bot[bot]