prusa3d/PrusaSlicer #5340: PrusaSlicer 2.2.0+ will not change extruder

ptrkueffner, 2020-11-29:
### Version

2.2.0+win64

### Operating system type + version

Microsoft Windows 10 Home Version 10.0.18363 Build 18363

### 3D printer brand / version + firmware version (if known)

Prusa MK3S + MMU2S

MK3S Firmware: 3.9.1-3518
MMU2S Firmware: 1.0.6-372

### Behavior

While setting up .3mf files for multi material prints:
 - Click the extruder drop down box to change the part extruder
 - Random parts will not close the drop down box and will not change the extruder in the plater window
     (see images)

 - After closing the PrusaSlicer, an error window pops up, see images

[image]

The central hexagon should change to extruder 1, but it's stuck on extruder three, and the extruder drop down won't close.

Error message:

[image]

#### Project File (.3MF) where problem occurs

[Handle Body.zip](https://github.com/prusa3d/PrusaSlicer/files/5613033/Handle.Body.zip)

bubnikv, 2020-11-30:
Did you compile PrusaSlicer yourself?
What build? With which wxWidgets version?
It looks like some development build with the FPS being shown in the right upper corner.
@YuSanka cannot reproduce your issue.

ptrkueffner, 2020-11-30:
I did not compile it myself - the FPS is from an NVIDIA gaming overlay that I've had running for all previous versions of PrusaSlicer.

The issue is random and sometimes doesn't affect a part.

I've attached another .3mf with more bodies as well as the folder with the wxWidgets and config

[Cave In Side 1.zip](https://github.com/prusa3d/PrusaSlicer/files/5619220/Cave.In.Side.1.zip)

[Alien.zip](https://github.com/prusa3d/PrusaSlicer/files/5619213/Alien.zip)

SIGSTACKFAULT, 2022-02-02:
Happening to me now. Manjaro Linux, got prusaslicer from pacman.

SIGSTACKFAULT, 2022-02-02:
Fixed by deleting `~/.config/PrusaSlicer`, and going through the setup again.

SIGSTACKFAULT, 2022-02-02:
If you're on windows and reading this, try looking in %appdata% for prusaslicer's settings? or just google it.

lukasmatena, 2022-02-03:
> Happening to me now

In what PrusaSlicer version is this happening now? The original report is from 2.2.0, can you reproduce it in 2.4.0? Thanks.

SIGSTACKFAULT, 2022-02-07:
@lukasmatena it was 2.4.1, from https://aur.archlinux.org/packages/prusa-slicer-git

closed 2026-09-29 by prusaslicer-bot[bot]