prusa3d/PrusaSlicer #5341: macOS Packaging Request

reedy, 2020-11-30:
Hello!

It's useful to have the current stable and the alpha/beta both installed for testing/comparison, but if you drag the app into Applications, it will want to overwrite the stable one (due to the naming clash).

I'm not very familiar with macOS packaging; but is it possible to do something with the names (and possibly whatever internal names) so that I can have multiple versions in the Applications folder, and as such, appear in Launchpad, without the beta/alpha files overwriting the stable release that is currently installed?

Thanks!

xarbit, 2020-11-30:
just rename it ti PrusaSlicer-beta and it won't overwrite ..

reedy, 2020-11-30:
Unfortunately, that doesn't seem to work. Hence filing the issue

If I use `PrusaSlicer-2.2.0+-202003211132.dmg` to first install the stable release (drag it into Applications), it appears. If I then use `PrusaSlicer-2.3.0-beta1+-202011281022.dmg` and drag it to Downloads, rename it (to PrusaSlicer-Beta) and then drag it into Applications.... Only the Beta shows in Launchpad (it replaces PrucaSlicer, if you immediately open launchpad you get to watch it be replaced/renamed). So it looks like something conflicts

[image]
[image]

macOS 10.15.7

xarbit, 2020-11-30:
yes, you are right, the app launcher does not show, understand.. 

You can find the other one here:

[image]

reedy, 2020-11-30:
Sure, I know you can find them there (as I've shown in my second screenshot).

But that fails on the usability front

xarbit, 2020-11-30:
sure, but that is not what you initially wrote ;-) .. I assume it's related to how Launchpad handles application bundle identifiers. 
This is something Apple decided to do so you don't have duplicate Apps in Launchpad .

reedy, 2020-11-30:
> sure, but that is not what you initially wrote ;-) .. I assume it's related to how Launchpad handles application bundle identifiers.

Well, I kinda did...

> I'm not very familiar with macOS packaging; but is it possible to do something with the names (and possibly whatever internal names) so that I can have multiple versions in the Applications folder, [...]

I did say `appear in Launchpad` ;)

I'd presume there is some way to make this happen... :)

bubnikv, 2020-12-07:
@xarbit 
> sure, but that is not what you initially wrote ;-) .. I assume it's related to how Launchpad handles application bundle identifiers.
> This is something Apple decided to do so you don't have duplicate Apps in Launchpad .

We would have to release the alphas, betas, various versions of PrusaSlicer as different applications to be able to install all of them into applications. We are not going to do that.

You may rename and copy multiple PrusaSlicer applications to desktop for example, that works.

Closing.

closed 2020-12-07 by bubnikv

reedy, 2020-12-08:
> We would have to release the alphas, betas, various versions of PrusaSlicer as different applications to be able to install all of them into applications. We are not going to do that.

Why not?

If your release building process is automated/scripted enough, changing it should be minimal extra work...

Why would I put random applications on my Desktop? Or anywhere else? That's not how macOS is generally designed in terms of workflow... Launchpad is there as the application launcher

reedy, 2020-12-08:
Digging around... You already do some variance for the betas... such as this from 30d7ef2cf911c89e36aa373794952fb947929cac, which seems a good time to do these sorts of packaging changes... https://github.com/prusa3d/PrusaSlicer/blob/d7412d4e54c9339300ac8ff5175c70278fd0d55e/src/slic3r/GUI/GUI_App.cpp#L666-L668

as presumably at some point you will change that back (even if in a branch)..

Is that manual or via a script?

From what I can see, the only real problem is where based on version.inc that `SLIC3R_APP_NAME` is used in various strings, you probably don't want "beta"/"alpha" appearing there in those cases, so a bit of refactoring to keep the generic name when used for display purposes inside the app (not that many usages), with a "DISPLAY_NAME" type variable that is literally always just `PrusaSlicer`, and then have the others change to `PrusaSlicer-Alpha`/`PrusaSlicer-Beta` or similar

reedy, 2020-12-08:
Is this a blanket no? Or would a PR be accepted? I don't really know what to read into your putting of a heart on my reply after your closing of the task.

If this was a more traditional FOSS-type project, I'd happily buy someone a beer/beverage of choice for implementing something that would be a usability improvement. Or donate to the project itself.

I'd presume I'm not the only person who might want to run multiple versions concurrently on macOS.

But as most of the devs are paid by Prusa, which is obviously a for profit company, I can't "donate" to Prusa, but I can (and have) spend money. Obviously this doesn't mean you default should do what I ask "just because I've spent money" etc.

The offer of buying a beer would still be on the table, as long as it doesn't violate any employment contracts etc if they're Prusa employed. Same for any other active volunteer dev on the project if they wanted to claim it.

As per my last comment, you're seemingly changing some of the strings anyway... The only difference would presumably you changing different strings in future... And maybe an extra string might need implementing that doesn't vary between alpha/beta/stable to be used for some display purposes

I can't see how it'd be that much work for a developer familiar with the project, and macOS development. I'm not a macOS developer, and I'm still relatively new to PrusaSlicer even on the user front.

And if you're doing it all manually, maybe some automation may help. Sure, I'm aware that automation sometimes is more time/effort than it's worth, vs just doing a small task manually every time...

bubnikv, 2020-12-09:
: Is this a blanket no? Or would a PR be accepted?

I cannot say. It depends on the amount of work required from our side (reviewing, merging, testing, maintaining) and on the complexity (testing and maintaining into the future).

Please understand that we are very busy. No matter how small this feature sounds to you, it adds to the overall load of the development and testing team and there are many issues requiring our attention, which IMHO are more important, for example the supports or some over-extrusion and gap fill issues.

reedy, 2020-12-09:
We're all busy, that's why things like this are useful. Just because I'm the first person to request something, doesn't mean I'm the only one. It's easier to have people help test things if they can be installed in a sane and useful/reliable way, rather than leaving applications all over the place.

One of the problems here is your first response of "We are not going to do that" came across as very final. Your second respons comes back as isn't quite as blunt/final. Does the PrusaSlicer project not have a Project Manager (or similar) that is responsible for deciding these sorts of things? Rather than someone that is closing a task then coming back with a vague answer and can't actually say whether they will or won't even accept a PR.

I've been involved in Open Source software (as a developer) for a good 15+ years at this point (please look at my contribution history). I understand working in public with the public dictating an amount of your work (look at my day job; we work a lot more in public than this). Communicating with random people on the internet is hard.

I understand having other priorities, but I also understand there is a difference between "this is something we don't ever want" through to "this is something we aren't going to work on ourselves, but we would accept a PR for" and then "yes, we'll do that immediately!" etc. If it's something that you'll never do, say that and close it. If it's something you might do later, comment to that effect, say you'd accept PR etc.

I never expected anyone to jump and do it immediately.

Unfortunately GitHub's issue tracker isn't great as tasks are either open or closed. There's no prioritisation unless you make those labels yourselvres etc.