prusa3d/PrusaSlicer #5342: Unable to build slicer with latest boost lib.

AdamNuclear, 2020-11-30:
Hi,

I am unable to build slicer 2.x.x with boost later than 1.72.0:

```
*  dev-libs/boost
      Latest version available: 1.74.0-r1
      Latest version installed: 1.72.0-r2
```

1.72.0 is ~1y old (atm) so you can expect 1.74.0 will be shipped to all distros soon.

```
FAILED: src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_LOG_DYN_LINK -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/media/data/portage_tmp/portage/media-gfx/prusa-slicer-2.3.0_alpha0/work/PrusaSlicer-version_2.3.0-alpha0/src -Isrc/platform -I/media/data/portage_tmp/portage/media-gfx/prusa-slicer-2.3.0_alpha0/work/PrusaSlicer-version_2.3.0-alpha0/src/clipper -I/media/data/portage_tmp/portage/media-gfx/prusa-slicer-2.3.0_alpha0/work/PrusaSlicer-version_2.3.0-alpha0/src/polypartition -isystem /usr/include/eigen3  -DNDEBUG -march=skylake -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder -fPIC   -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -std=gnu++17 -MD -MT src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o -MF src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o.d -o src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o -c /media/data/portage_tmp/portage/media-gfx/prusa-slicer-2.3.0_alpha0/work/PrusaSlicer-version_2.3.0-alpha0/src/admesh/stlinit.cpp
/media/data/portage_tmp/portage/media-gfx/prusa-slicer-2.3.0_alpha0/work/PrusaSlicer-version_2.3.0-alpha0/src/admesh/stlinit.cpp:31:10: fatal error: boost/detail/endian.hpp: No such file or directory
   31 | #include <boost/detail/endian.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
The expected file was removed by this commit:

https://github.com/boostorg/predef/commit/aa6e232bf170ad8b856aff9e7c70334f77441c7f

When I downgraded the boost from 1.74 to 1.72 it compiles without any issue.

OS: Gentoo
GCC: gcc version 9.3.0 (Gentoo 9.3.0-r1 p3)

martin-kokos, 2020-11-30:
The header file location has been updated here: https://github.com/prusa3d/PrusaSlicer/commit/3a194f91871684b5fc8c6c78300b87e0ee1cb610
Can you check latest master?

rtyr, 2021-01-28:
No response, likely not issue anymore. Closing.

closed 2021-01-28 by rtyr