prusa3d/PrusaSlicer #5344: PrusaSlicer crashes upon slicing with specific combination of print settings and part geometry

protonenfalter, 2020-11-30:
### Version
PrusaSlicer 2.3.0-alpha4+linux-x64

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S, FW 3.9.1-3518

### Behavior
The attached STL 'Stütze-R8.stl' loads just fine into PrusaSlicer. With standard settings it will also slice as it is supposed to.

However, with slightly altered slice settings ('0.30mm Generisch@MK3S') , PrusaSlicer crashes without warning when 'Slice Now' is pressed.

#### Project File (.3MF) where problem occurs
[Stütze-R8.zip](https://github.com/prusa3d/PrusaSlicer/files/5616060/Stutze-R8.zip)

bubnikv, 2020-11-30:
please test with beta1. It should work.

protonenfalter, 2020-11-30:
Indeed, with the Beta-1 of PrucaSlicer the problem doesn't occur. The part slices as expected.

Thank you!

closed 2020-11-30 by protonenfalter