prusa3d/PrusaSlicer #5345: G-Code generated starts not with "Start G-Code" inputbox, M80 command Marlin not working /not supported

Zuckme, 2020-11-30:
### Version
2.20

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 5

### Behavior
* _Describe the problem_

Set a Start G-Code
[image]

* _Expected Results_

Gcode generated:
`; generated by PrusaSlicer 2.2.0+win64 on 2020-11-30 at 18:25:21 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.40mm
; first layer extrusion width = 0.70mm

**;FIRST LINE START G-CODE
M80 ; power on
G28 ; home all axes
; ENABLE_LEVELING_FADE_HEIGHT
M420 Z3
.....**
M73 P0 R57
M73 Q0 S68
M201 X500 Y500 Z100 E5000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z10 E50 ; sets maximum feedrates, mm/sec
M204 P500 R1000 T500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.30 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M190 S75 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature

* _Actual Results_

Gcode generated:
`; generated by PrusaSlicer 2.2.0+win64 on 2020-11-30 at 18:25:21 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.40mm
; first layer extrusion width = 0.70mm

M73 P0 R57
M73 Q0 S68
M201 X500 Y500 Z100 E5000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z10 E50 ; sets maximum feedrates, mm/sec
M204 P500 R1000 T500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.30 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M190 S75 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
**;FIRST LINE START G-CODE
M80 ; power on
G28 ; home all axes
; ENABLE_LEVELING_FADE_HEIGHT
M420 Z3
...**

As you can see in the PrusaSlicer generated G-CODE , the "Start G-Code" block is not the first, even a M190 and a M104 are placed before the `;FIRST LINE START G-CODE` so in my case the printer quit the printing job and goes in error state.

My understanding of the input box "Start G-Code" to be placed before everything.

As a stop gap solution you should add a checkbox in general printer setting for M80/M81 support.

Thanks.

bubnikv, 2020-12-01:
> As you can see in the PrusaSlicer generated G-CODE , the "Start G-Code" block is not the first, even a M190 and a M104 are placed before the ;FIRST LINE START G-CODE so in my case the printer quit the printing job and goes in error state.

Why? What is going wrong? What does M80?
How come the other Ender 3 owners do not complain?

Zuckme, 2020-12-01:
>  What does M80?

[M80](https://marlinfw.org/docs/gcode/M080.html): Turn power supply [PSU] on
[M81](https://marlinfw.org/docs/gcode/M081.html): Turn power supply [PSU] off

> Why? What is going wrong? 

For users who would like to turn the PSU at the beginning of the printing and then turn it off when the print is done, Prusaslicer can not do the job.
Prusa slicer generate a M190 and a M104 before the user can turn on the power supply ( "Start G-Code" block). Marlin detects those commands but without the power supply on quits the printing job and goes in error state.
In other words you can not use the heaters if there is no power for them.

> How come the other Ender 3 owners do not complain?

It has nothing to do with a Ender 3 or a Ender 5.
In other words,

1. "Start G-Code" is not the beginning of the generated g-code, but after a M190 and a M104, this is confusing and misleading.
2. In Prusaslicer it is not possible to turn on a printer before the printing job starts (M80 not supported)

Please help your users to save energy!

rtyr, 2020-12-01:
It should be doable in 2.3.0-beta. 
Try this:

```
M80 ; power on
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 ; home all axes
; ENABLE_LEVELING_FADE_HEIGHT
M420 Z3
M201 X500 Y500 Z100 E5000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z10 E50 ; sets maximum feedrates, mm/sec
....
....
```

As for the machine limits (M20X commands) you have more options.
[image]

`Emit to G-code` - M20X commands will be placed to the beginning of the gcode (before `M80`)

`Use for time estimate` - probably the best option in this case. PS will not emit any M20X commands to the gcode (but the values will be used for time estimate). You can add those commands manually to your start g-code (if needed).

`Ignore` - Machine limits are ignored (time estimate may not be accurate)

https://github.com/prusa3d/PrusaSlicer/releases

bubnikv, 2020-12-01:
Also you would have to place the print bed and preheat commands into your
custom G-code.
I am not sure however that all the G-code will disappear from before the
custom start G-code. Fan off may be still left.

I still wonder what is your hardware configuration: What does this power
off G-code do? Who interprets it? How is your hardware powered? Marlin
board powered from different power supply than the rest of the printer?

Zuckme, 2020-12-01:
> You can achieve similar result in 2.2.0 as well (by switching g-code flavor to `reprap/sprinter`).

Unfortunately no, see below.

`; generated by PrusaSlicer 2.2.0+win64 on 2020-12-01 at 11:22:47 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.40mm
; first layer extrusion width = 0.70mm

M73 P0 R15
M107
M190 S75 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
;FIRST LINE START G-CODE
M80 ; power on `

> It should be doable in 2.3.0-beta.

How can I upgrade without losing all my 2.2 configurations?

>  I still wonder what is your hardware configuration: What does this power off G-code do? Who interprets it? How is your hardware powered? Marlin board powered from different power supply than the rest of the printer?

My hardware is what I hope all future 3D Printer will/should have. The controller board is powered by a 5V source independent by the main big 12V-24V PSU (for the heaters/stepper you name it). This can be done easily for example by any raspberry (octoprint host) USB 5V cable connected to the printer board, so nothing NASA complicated. I can send gcode to the board WITHOUT the main PSU on. In Marlin you can configure that at startup with `#define PS_DEFAULT_OFF` , see [here](https://marlinfw.org/docs/configuration/configuration.html#power-supply) for details.

The big PSU 12V/24V can be turned off and on by Marlin with M80/M81, %V it always there (raspi always on). Normally you do not need any 12V/24V if you are not printing (fan noise and energy reduction), this is the magic behind everything.

How you get the main PSU responding to M80/M81 is a little challenging, normally the solution is to connect a relay controlled by the board in series to the PSU mains. Tons of how to in internet.

In my particular case I have a [SKR mini E3 V3](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3) connected with a 
[BIGTREETECH Relay V1.2](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2).

The BIGTREETECH manager in their immense wisdom have dropped the M80 support in the new relay board V1.2 (V1.0 could do that.... ), so I had to [hack](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/7#issuecomment-735447011) it.

DO NOT THINK I AM A SINGLE PARTICULAR HACKER WITH A VERY SPECIFIC NEED!
It is not the case, I hacked that board because it was not doing his job properly as it should.

If you are asking me so much about M80/M81 and by looking at your PrusaSlicer I assume the Prusa printers do not support a M80/M81, please give a call to the rnd guys and let them do thier magic for future models.
Meanwhile just add a "Remote PSU Contol" in the expert profile and two checkboxes, something like that:

Remote PSU Contol
- [ ] Turn PSU on (send M80) before printing
- [ ] Turn PSU off (send M81) after printing

rtyr, 2020-12-01:
>Unfortunately no, see below.

You did not use the start gcode I suggested (you have to place the print bed and preheat commands into your
custom g-code.)... Reprap/sprinter flavor is just to avoid emitting machine limits in 2.2.0.

Works fine in 2.2.0

[image]

Zuckme, 2020-12-01:
> You did not use the start gcode I suggested 

True I wanted just to see where the `;FIRST LINE START G-CODE` where placed, I will try it later.
It is an ugly workaroud btw, so I have to tip all the temperatures by hand and they are not automagically translated in G-CODE as before from my profiles....
But eh... better than nothing.
Thanks in advance I let you know when I have tested it.

rtyr, 2020-12-01:
> 
> It is an ugly workaroud btw, so I have to tip all the temperatures by hand and they are not automagically translated in G-CODE as before from my profiles....
> But eh... better than nothing.

You can see placeholders in my example start g-code....so the temperatures will be automatically translated into the g-code of course.

[image]

Zuckme, 2020-12-01:
Ahhh okay sorry I interpreted those `[...]` as "put what you need"....
Now we are talking....
Thanks!

BTW: How funny is [this ](https://github.com/slic3r/Slic3r/issues/1038). I suggest to implement two checkboxes for the PSU control. Easy.

al3ph, 2020-12-19:
Yeah this change is causing me issues as well, I use a custom preheat macro (klipper), which soaks the bed based off the current bed temp->target bed temp, this no longer works as the bed is already at the set temperature by the time Start G-code is called.
This is likely to cause issues for a lot of klipper users. Start G-Code should be the first command to be executed, or a new Start G-code property needs to be added which is "really" the first command in the gcode file.

Cheers

al3ph, 2020-12-19:
Adding 
```
M104 [first_layer_temperature] ; hack
M140 [first_layer_bed_temperature] ; hack
M104 0 ; hack
M140 0 ; hack
```
To custom gcode works, but this is going to trip up a few people, I assume previously it just looked for the variables being used in the custom gcode..

cnf, 2021-01-15:
I have this problem as well, I am using klipper and my start gcode looks like this:
`PRINT_START EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]`
it sets all the temps i need in a macro on klipper

Let us decide what gcode gets included, please....

Arcturuss, 2021-02-15:
apparently it was sorted out in SuperSlicer with 'klipper' as additional gcode flavor https://github.com/supermerill/SuperSlicer/issues/232

rtyr, 2021-02-25:
So in general, this issue is about the same thing as https://github.com/prusa3d/PrusaSlicer/issues/2420. Closing this one as a duplicate.

closed 2021-02-25 by rtyr

tarsonis123, 2021-03-02:
Sorry for pushing this again. Even if the flavour is set to Klipper the filament Gcode nevertheless auto generates temperature gcodes to the file. How to disable this? This behavior is kind of irritating  

```
;TYPE:Custom
START_PRINT T_TOOL=205 T_BED=60 ; Klipper Start code Macro
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament Gcode
M104 S205 ; set temperature
M109 S205 ; set temperature and wait for it to be reached
M190 S60 ; set bed temperature and wait for it to be reached
```