prusa3d/PrusaSlicer #5348: Unhandled HTML in Astrobox connection error message

blue-hound, 2020-11-30:
### Version
PrusaSlicer-2.3.0-beta1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S MMU2ZS

### Behavior
Unhandled HTML in Astrobox setup
```Could not connect to AstroBox: HTTP 404: <html>
<head><title>404 Not Found</title></head>
<body bgcolor="white">
<center><h1>404 Not Found</h1></center>
<hr><center>nginx</center>
</body>
</html>
Note: AstroBox version at least 1.1.0 is required.```

I do not have Astrobox, I stumbled on this while testing the FlashAir connection.

blue-hound, 2020-11-30:
Actually, this occurs in all the connection errors.

[image]

blue-hound, 2020-12-04:
I just received a similar error in 2.2.0 as well when an upload failed to a FlashAir card. I clicked it away before I thought of taking a screen grab, but this appears to be present in earlier versions as well.

bubnikv, 2020-12-07:
What is the issue here? That the HTML block is not formatted? Or that the error shows up?

blue-hound, 2021-01-09:
As per the title, the unhandled HTML is the issue. The errors were displayed appropriately.

bubnikv, 2021-01-11:
> As per the title, the unhandled HTML is the issue. The errors were displayed appropriately.

I don't understand> What do you mean with "unhandled HTML is the issue?"

blue-hound, 2021-01-11:
HTML appears in the error message unformatted. It should be plain text.

blue-hound, 2021-01-11:
Appears to be fixed in 2.3.0.

closed 2021-01-11 by blue-hound