prusa3d/PrusaSlicer #5349: Head crash on completion with sequential printing

stixpjr, 2020-12-01:
### Version
Version: 2.2.0
Build: PrusaSlicer-2.2.0+UNKNOWN
Operating System: Unix
System Architecture: 64 bit
System Version: NetBSD 9.1_STABLE amd64
Total RAM size [MB]: 17,127MB

### Operating system type + version
System Architecture: 64 bit
System Version: NetBSD 9.1_STABLE amd64

### 3D printer brand / version + firmware version (if known)
Prusa Mini
4.2.1-final+2072

### Behaviour
Upon completion, the head "parks", but this parking does not take into account other objects on the platter in sequential printing mode ("Complete individual objects"). Instead of attempting to change the object print order or arrangement (which is hard), surely just moving the Z-axis to the maximum height of existing objects prior to shifting X&Y axes would be sufficient?

[image]

#### Project File (.3MF) where problem occurs
[PrusaSlicerHeadCrash.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5619584/PrusaSlicerHeadCrash.3mf.gz)

bubnikv, 2020-12-01:
e62bb4a7c7bc93c62ffb05a848fd63f862b5b633 implements
    In addition to an existing "layer_z" placeholder variable,
    a new "max_layer_z" has been added to custom G-code sections accumulating
    maximum of "layer_z" over the print. This value may be used for example
    in the end G-code of a sequential print to lift the print head
    up before travel over the whole print bed.

Now all the layer_z occurences in the custom end G-code need to be replaced with max_layer_z.

```
G1 E-1 F2100 ; retract
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)}{endif} F720 ; Move print head up
G1 X178 Y178 F4200 ; park print head
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} F720 ; Move print head further up
G4 ; wait
```

@rtyr would you please take care of that? Also the new config bundle needs to be made compatible with beta2, as beta1 will not understand the new max_layer_z placeholder.

closed 2020-12-01 by bubnikv

stixpjr, 2020-12-01:
Perfect! Thanks!