prusa3d/PrusaSlicer #5350: "Add instance" and "Remove instance" hotkeys don't work when object is selected via de object browser

jaapp, 2020-12-01:
### Version
Prusaslicer 2.4.0-beta1

### Operating system type + version
MacOs 10.14.6

### Behavior

The + and - hotkeys don't work in the object-browser

* Start Prusaslicer
* Add STL to buildplate
* Select object via object browser (click on object so it's highlighted blue)
[image]
* Press "add instance" or "remove instance" hotkey on keyboard (+ and - keys)
* No new instance is made / instances can't be removed this way
* Clicking on the corresponding buttons *does* work
[image]
* When you select an object by selecting them on the buildplate the hotkeys also *do* work.

_Is this a new feature request?_
No

fe60, 2020-12-07:
>     Also, when you select an object by selecting them on the buildplate the hotkeys _do_ work.
To me it seems to be related to the location of the mouse pointer, not to the way an object has been selected.
1. Select an object in the right panel/object list
2. Keep mouse pointer over right panel
3. use `+` / `-` to add/remove an instance
=> nothing happens
4. Move mouse pointer back into 3D view (don't click, just move)
5. use `+` / `-` to add/remove an instance
=> works

Tested with PrusaSlicer-2.3.0-beta2+win64-202012051115

jaapp, 2020-12-07:
I had to test @fe60 his remark and yes - it is related to the mouse pointer! So weird. None of the other hotkeys seem to have this limitation. It is very unintuitive. And the effect is that you can't use your keyboard to do some quick instance creation without involving the mouse.

jaapp, 2020-12-08:
Great to see this is getting addressed in the next release. This means that the most important hotkeys are now supported by the object list.

I've looked at the changes in the source code and this prompted me to test some other hotkeys. These are all the hotkeys that don't work when the object list is selected:

From the top toolbar:
* `A` - Arrange all
* `Shift+A` - Arrange all selected

From the toolbar on the left:
* `M` - Move
* `S` - Scale
* `R` - Rotate
* `F` - Place on face
* `C` - Cut
* `L` - Paint on support
* `P` - Seam painting

The hotkeys for the left-toolbar all require the user to do some object manipulation with the mouse in the 3D workspace so I guess there is a reason for these not to work outside of this context - although its arguably not the most consistent user experience.

The same can't be said for the top toolbar. Hotkeys `A` and `Shift+A` don't require any manual object manipulation. It would be great to add these too. Plus that would also mean that the entire top-toolbar is now supported by the object list.

Thanks for taking the time to read my feedback!

bubnikv, 2020-12-12:
Implemented for Windows in PrusaSlicer 2.3.0-beta3.
On OSX the +- accelerators do not work reliably with a localized keyboard.

jaapp, 2020-12-14:
@bubnikv I've tried beta3 and the +- hotkeys don't work on a standard keyboard. 

I have a mid-2014 13-inch Macbook Pro with a (US) English International Layout keyboard layout.

bubnikv, 2021-10-04:
We are limited on OSX with the wxWidgets UI wrapper library. We install an accelerator table, but for some reason it seems the support of accelerator tables in wxWidgets is not right. We would likely have to fix wxWidgets.

> I wonder if the right keycodes to catch should be WXK_ADD and WXK_SUBTRACT

@YuSanka tried that. It works for +/- on numerical keyboard only, which the macs don't offer by default.

closed 2026-09-29 by prusaslicer-bot[bot]