prusa3d/PrusaSlicer #5353: PrusaSlicer 2.3 beta 1 - Open new instance not available on macOS Big Sur

drelich, 2020-12-01:
### Version
2.3.0 Beta 1

### Operating system type + version
macOS Big Sur 11.0.1

### Behavior
* Open new instance greyed out/not available

[image]

BryanSmithDev, 2020-12-02:
Someone else may know better and correct me, but I think this is related more to MacOS and less about PrusaSlicer. I believe they had to specifically limit MacOS PrusaSlicer to only a single instance because of a limit in how the Mac operating system works.

drelich, 2020-12-02:
All four alpha versions had that feature enabled and 100% working on macOS. Only the Beta 1 does not. Cheers.

BryanSmithDev, 2020-12-02:
Apologies @drelich, I was incorrectly recalling an issue specifically relating to the dock and Finder mentioned below, but that does not apply in all cases, including yours I believe.

> OSX specific: The OSX Finder and Dock only support a single application instance. Thus when double clicking on a PrusaSlicer icon, the existing PrusaSlicer instance is always activated independent of the "single instance" settings, and drag & dropping a file onto a PrusaSlicer icon in Finder or Dock opens the file in some random, likely the first instance of PrusaSlicer. However when starting PrusaSlicer.app/Contents/MacOS/PrusaSlicer from terminal, the "single instance" preferences settings is respected. We are aware that the approach of starting multiple instances of the same application on OSX is neither recommended nor supported by the OSX framework, however for technical reasons this was the quickest way to deliver multiple PrusaSlicer windows and the stand-alone G-code viewer.

Maybe someone else that is more familiar will respond to this issue. Until then, since much has changed during the alpha and beta releases, double-check that "Single instance mode" is not checked under Preferences as that will inherently cause the "Open new instance" option to be greyed out.

closed 2020-12-02 by drelich

MartinMajewski, 2020-12-08:
I was about to report the same issue for beta 2. It is quite unfortunate, how (dev-)version changes lose all the settings and profiles every time. By all the time has passed since the first alphas, I already forgot about the settings option that has to be unchecked for the multi-instance behavior.

I will open a new ticket asking for some UX improvements related to this.