prusa3d/PrusaSlicer #5354: PrusaSlicer-2.2.0 crashes on startup when removable media is not mounted

oh2gqc, 2020-12-01:
### Version
_Version of PrusaSlicer used goes here_
2.2.0+linux-x64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
XUbuntu 18.04.5 LTS (Bionic Beaver)

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
N/A

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
N/A

### Behavior
* _Describe the problem_
Program ends with an error message.
(prusa-slicer:13962): GLib-GObject-CRITICAL **: 16:37:33.170: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
what(): boost::filesystem::space: No such file or directory: "/media/Backup"
Aborted (core dumped)

This only happens if removable media is not mounted on /media/Backup.
If mounted program starts normally.

* _Steps needed to reproduce the problem_
Just start pgm "./PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage"
when removable media is unmounted.

 * _If this is a command-line slicing issue, include the options used_
No

* _Expected Results_
Program startup.

* _Actual Results_
Crash with error message(s) See above.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

kocikdav, 2020-12-02:
Hello, thank you for reporting. I suspect the crash is happening when inspecting possible removable drives. Please would you try replicate with this build? If crash still happens, more detailed log should be printed.
https://www.dropbox.com/s/fyfd8dqfdrqg299/PrusaSlicer-2.3.0-beta1%2B45-linux-x64-asan-g921bd59-202012021213.AppImage?dl=0

oh2gqc, 2020-12-02:
Hi, thank you for the build you provided.

I tested it with removable media both  mounted and unmounted.

I'm glad to report that the program did not crash anymore.
It started OK in both cases.

BR, Kari

P.S. By "OK", I mean that it did not crash on startup. There are however other messages printed out ( attached )
[PrusaSlicer.txt](https://github.com/prusa3d/PrusaSlicer/files/5630126/PrusaSlicer.txt)

kocikdav, 2020-12-02:
Thank you for testing. im glad you can run PrusaSlicer now. This fix will be part of the next beta release. Im closing this issue now (I will still investigate the attached log).

oh2gqc, 2020-12-03:
Thank you for  fixing the issue in such a short time!

BR, Kari