prusa3d/PrusaSlicer #5356: Prusaslicer can't upload to Octoprint 1.5

henry-feldman, 2020-12-01:
### Version
2.2.0+

### Operating system type + version
MacOS Big Sur 11.0.1
PrusaSlicer
Version:   2.2.0+
Build:     PrusaSlicer-2.2.0+-202003211132

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.16 (Build 20B29)
Total RAM size [MB]: 68,719MB
OpenGL installation
GL version:   2.1 ATI-4.0.47
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 5600M OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
I upgraded to Octoprint 1.5 this morning. Since doing so the upload button in prusaslicer reports a 404 error as follows:
Could not connect to OctoPrint: HTTP 404: <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL was not found on this server.</p>
</body></html>
Note: OctoPrint version at least 1.1.0 is required.

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs

[LED hex grid.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5625731/LED.hex.grid.3mf.zip)

bubnikv, 2020-12-02:
> <title>404 Not Found</title> Not Found

>  Since doing so the upload button in prusaslicer reports a 404 error as
follows:
> Could not connect to OctoPrint: HTTP 404:

Which button?

The "Test" button to test the connection?

[image: image.png]

Or the G-code upload button?

[image: image.png]

henry-feldman, 2020-12-02:
> Since doing so the upload button in prusaslicer reports a 404 error as
> follows:
> Could not connect to OctoPrint: HTTP 404:
> Which button? The "Test" button to test the connection? [image: image.png] [...]

I noticed the issue when using the regular upload button (the tiny G with an arrow) however the error message above was via the test button. I am using zero-conf (i.e. Bonjour) to connect (so not a numeric IP, but the .local method, so it is to octopi.local). Now my browser to the same address works fine (and in fact after encountering the error just used the export g-code and uploaded via the browser like we did in the old days)
[image]

henry-feldman, 2020-12-02:
OK, so based on the same discussion over on the prusaslicer side the issue appears to be using zeroconf. When I switched the IP address to my LAN IP address for the Raspberry Pi (192.168.1.55) it worked perfectly. So I am guessing this must somehow be related to the zeroconf issue others have mentioned. Now the mac has no problem resolving the .local address via the browser.

bubnikv, 2020-12-03:
Is your issue the same as https://github.com/prusa3d/PrusaSlicer/issues/1781 ?

henry-feldman, 2020-12-04:
> Is your issue the same as #1781 ?

Not exactly. That was due to (at least my quick read) the windows user not having the bonjour resolver installed, but macOS natively includes bonjour (in fact I believe Bonjour is an apple TM, rather than ZeroConf is the generic version). There is a lively thread over in the [octoprint [](forum](https://community.octoprint.org/t/cannot-connect-with-local-name-only-with-ip-address-after-1-5-0-update/26907/42)) where she thinks she has an idea of the root cause (has something to do with the discovery service)

bubnikv, 2020-12-04:
For reference:
https://community.octoprint.org/t/cannot-connect-with-local-name-only-with-ip-address-after-1-5-0-update/26907/44
https://community.octoprint.org/t/octoprint-1-5-throws-a-404-in-prusaslicer/26941/7
https://github.com/OctoPrint/OctoPrint/issues/3852

It looks it is an issue of OctoPrint 1.5.0 (or OctoPI 1.5.0 image?) Bonjour network discovery issues.
Downgrading to OctoPrint 1.4.0 apparently solves the issue.

henry-feldman, 2020-12-04:
Octoprint has released a patch for this issue:

https://octoprint.org/blog/2020/12/04/new-release-1.5.1/?utm_medium=announcements&utm_source=octoprint&utm_content=1.5.0

closed 2020-12-04 by henry-feldman

ntmaster26, 2020-12-04:
Unsubscribe 

Closed #5356.