prusa3d/PrusaSlicer #5358: manifold check regression since 2.2.0

expipiplus1, 2020-12-02:
### Version

30d7ef2cf911c89e36aa373794952fb947929cac

### Operating system type + version

- GNU/Linux, NixOS

### Behavior

- Open the attached project in 2.2.0
- observe that you can observe the hollowed and drilled model without error
- Open in 30d7ef2cf911c89e36aa373794952fb947929cac
- Observe that when trying to preview the model the application fails with `Mesh not watertight`
- Observe that one can see the hollowed model if one removes the holes!

It might be that this model really is non-manifold, nevertheless it's a shame that what used to work does not any more!

#### Project File (.3MF) where problem occurs

https://gist.github.com/a3ba4d1cbcf732d77538304de0a7a667

HanClinto, 2021-02-02:
I'm getting this error very consistently.  I can hollow a model and generate supports without problem, but if I drill any holes into the model, I get an error with `Mesh not watertight` and I am unable to proceed.  This makes mesh hollowing largely useless if I'm not able to drill holes into the model.

bubnikv, 2021-02-02:
Will it help to fix the model using netfabb?
https://service.netfabb.com

bubnikv, 2021-09-22:
No response. We improved the hole drilling in PrusaSlicer 2.4.0-alpha1. Holes are being drilled if they do not intersect with non-manifoldity.

closed 2021-09-22 by bubnikv