prusa3d/PrusaSlicer #5359: Model with holes in the base does not slice properly in "Vase Mode"

PaulDC, 2020-12-02:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

PaulDC, 2020-12-02:
Prusa Slicer 2.3.0 beta1
Windows 10 professional
Prusa Mini

When slicing a model with holes in the base in Vase Mode the holes get filled in. STL file attached together with screen shots in a zip file.
[PrusaSlicer_2_3_0_beta1_VaseModeProblems.zip](https://github.com/prusa3d/PrusaSlicer/files/5627991/PrusaSlicer_2_3_0_beta1_VaseModeProblems.zip)

adrianobarrosmba, 2020-12-05:
Same problem here

bubnikv, 2020-12-09:
@FidelCapo narrowed it more closely to
5ff6f30 from 202008210958 - ok
f6acd49 from 202008261352 - not ok

bubnikv, 2020-12-12:
Fixed in PrusaSlicer 2.3.0-beta3

closed 2020-12-12 by bubnikv