prusa3d/PrusaSlicer #5361: Prusaslicer directory does not see STL file

lmmohr4, 2020-12-02:
### Version
_Version of PrusaSlicer used goes here_2.2.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Windows10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_Prusa I3 MK3

### Behavior
* _Describe the problem_DOES NOT FIND STL FILE IN CORRECT DIRECTORY WHERE I PLACED IT. NEWLY OPENED PRUSASLICER INSTANCE.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_NO

#### Project File (.PLA) where problem occu        -
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-12-02:
> DOES NOT FIND STL FILE IN CORRECT DIRECTORY WHERE I PLACED IT. NEWLY
OPENED PRUSASLICER INSTANCE.

I am sorry, but I cannot understand what you mean.

rtyr, 2020-12-03:
I would guess that you either using incorrect way to open it (Open project) or the file does not have correct extension (*.stl).

[image]

lmmohr4, 2020-12-03:
I placed the STL file in a known directory location with File Manager then
opened Prusaslicer. When I do as you show, the file is not shown. I have
tried numerous file manager programs. They all show the STL file as being
there. Prusaslicer does not. Slice3r sees it. Prusaslicer 2.2.0 based on
Slice3r. Please escalate.

FidelCapo, 2020-12-04:
Can you drag and drop this stl file into PrusaSlicer 3Dscene?

lmmohr4, 2020-12-04:
Sent from Mail for Windows 10 From: HurkaFSent: Friday, December 4, 2020 4:43 AMTo: prusa3d/PrusaSlicerCc: lmmohr4; AuthorSubject: Re: [prusa3d/PrusaSlicer] Prusaslicer directory does not see STL file (#5361) Can you drag and drop this stl file into PrusaSlicer 3Dscene?

rtyr, 2020-12-07:
Something went wrong with your last message..

Can you drag and drop this stl file into PrusaSlicer 3Dscene? Does it work? Any screenshots or additional info?

lmmohr4, 2020-12-08:
NO FILE ATTACHED. HOW DO I DO ANYTHING? Sent from Mail for Windows 10 From: rtyrSent: Monday, December 7, 2020 5:35 PMTo: prusa3d/PrusaSlicerCc: lmmohr4; AuthorSubject: Re: [prusa3d/PrusaSlicer] Prusaslicer directory does not see STL file (#5361) Something went wrong with your last message..Can you drag and drop this stl file into PrusaSlicer 3Dscene? Does it work? Any screenshots or additional info?

rtyr, 2020-12-14:
Look at this link in web browser to see what are you posting https://github.com/prusa3d/PrusaSlicer/issues/5361.

rtyr, 2021-07-23:
No response. Closing.

closed 2021-07-23 by rtyr