prusa3d/PrusaSlicer #5365: Random head parking on PS 2.3 beta 1

morphias2004, 2020-12-03:
### Version
2.3.0-beta1+win64

### Operating system type + version
Windows 10 Pro x64 19042.662 with experience pack 120.2212.551.0

### 3D printer brand / version + firmware version (if known)
Stock Prusa MK3S running FW 3.9.1.3518

### Behavior

At random points during the print, the head will go and park and the bed with retreat all the way to the back. The print will then resume where it left off.

The print is circular (a funnel) and the point in the circle where this happens appears to be the same each time, but the layer height is random. e.g. no apparent pattern like every 60 layers.

Whilst the head is parked, it oozes and when the print resumes, the ooze gets dragged across the print, leaving random deposits, which in turn produce flaws in the print.

These movements cannot be seen in the GCode viewer.

Further to this, When the printer does the gap infill between the inner and outer layers, instead of starting and stopping at the seam like all of the perimeters, the head over shoots the start of the seam, daps the prints surface and then returns to the start of the seam before doing the gap infill.

This also results in deposits on the print surface, creating flaws in the print and also not filling the first 15mm of the gap properly.

These movements can be seen in the GCode viewer.

If not for these two issues, this print would have come out completely flawless.

I have uploaded a video to YouTube showing both issues. The description has the time entries for where they occur.

https://youtu.be/IBGHnGUcXes

https://youtu.be/GYM1lCVk3as

#### Project File (.3MF) where problem occurs

3MF, STL and GCode in attached ZIP file

[Funnel.zip](https://github.com/prusa3d/PrusaSlicer/files/5633887/Funnel.zip)

rtyr, 2020-12-03:
**First issue**
I don't think it's related to the PrusaSlicer. Possibly false crash detection or some problem with SD card, communication error with Octoprint, etc.

**Second issue**
It looks like standard wipe movement. You can disable it in Printer settings - Extruder (Wipe while retracting).

Area5142, 2020-12-03:
**First issue**
I have false "crash detection" on both my MK3S printers as shown on the video - always on the y-axis.
The detection algorithm in the firmware is way to sensitive and causes the position reset and ooze blobs on the model.

I turn off crash detection to "fix" the problem. This has never caused me problems and until now I have not had any real crashes.

bubnikv, 2020-12-03:
If you think it is a firmware issue, then please post an issue at
https://github.com/prusa3d/prusa-firmware/issues
If you believe you have a hardware issue, then please call our support.

morphias2004, 2020-12-06:
OK.

After another couple of days of printing, I'm convinced the 'parking' is false crash detection without the printing actually saying a crash was detected.

This makes it a firmware issue and I'll open a case on the firmware site.

As for the second issue of the 'odd' movements on gap infill, I looked closely at the finished funnel and see the seam on the outside is one perfect line, but the seam inside steps back and forth. This is the cause of these movements.

Why the slicer has done the internal seam this way is beyond me and I feel it is a slicer issue. The seams were set to 'Alligned' in the slicer.

It's hard to take detailed pics of Prusa Orange, but here is the best I can do. I have (badly) drawn a line to the right of the seams to show the path they take.

[image]

[image]

[image]

morphias2004, 2020-12-12:
I have confirmed turning off crash detection stops the random head parking. I'll add the the case mentioned above in the Firmware Github.

The issue of the wonky seam alignment that causes scarring and poor gap fill between external perimeters remains in Beta 2:

Here is the same file sliced in Beta 2

Layer 142

[image]

Layer 143

[image]

Layer 144

[image]

Layer 145

[image]

SachCZ, 2023-12-19:
The original issue turned out to be a firmware issue. The other issue does not seem to be present anymore. If you think otherwise, please open a new issue.
Combining multiple issues makes it hard to keep track of them and complicates addressing them.

closed 2023-12-19 by SachCZ