prusa3d/PrusaSlicer #5367: Excessive support material above the model in 2.3.0-beta1

HiLivin, 2020-12-03:
### Version
2.3.0-beta1+win64

### Operating system type + version
Windows 7 64-bit

### 3D printer brand / version + firmware version (if known)
Ender-3 Pro

### Behavior
Excessive _support material_ and _support material interface_ is generated above the top layer of the model.
The "Support on build plate only" option is enabled.

#### Project File (.3MF) where problem occurs
[wave_disk.zip](https://github.com/prusa3d/PrusaSlicer/files/5636046/wave_disk.zip)
[image]

Hades32, 2020-12-10:
Can confirm on Win10. Set "overhang support" to 55° but still got similar artifacts.

HiLivin, 2021-09-04:
The issue seems resolved in 2.4.0-alpha1 with "Snug" support style chosen.

bubnikv, 2021-09-06:
cool, closing.

closed 2021-09-06 by bubnikv