prusa3d/PrusaSlicer #5368: User unable to add new material SPLA

Sugarpuffdaddy, 2020-12-03:
### Version
Version: 2.2.0

### Windows 10

### Prusa MK3s

User unable to add new material like SPLA

https://www.amazon.co.uk/gp/product/B089NLGMLF/ref=ox_sc_act_title_1?smid=A2YZ3BEOCEHX4S&psc=1

[image]

rtyr, 2020-12-03:
Please provide more detailed description of your issue. I don't understand what do you mean.

Do you mean filament type? You can add custom filament type.
[image]

Sugarpuffdaddy, 2020-12-03:
Are you using PrusaSlicer? I am an am unable to find where I can add a custom filament type. 

Please can you assist?

I

rtyr, 2020-12-03:
Just type your custom filament type and then save the filament profile.

[image]

If you are not sure, how to create new filament profile in general, please contact our tech-support.

closed 2020-12-03 by Sugarpuffdaddy