prusa3d/PrusaSlicer #5369: Support Modifier / Part Manipulation / Position / Z and Y axis are swapped

Nanodiver38, 2020-12-03:
PrusaSlicer-2.2.0+-2-2--3211132 based on Slic3r

macOS Catalina
Version 10.15.7

Prusa / MK3S + firmware version 3.9.3 
No modifications / Issue not with printer

### Behavior
When manipulating the position of a support enforcer (cylindrical in my case) the Y-position seems to be adjusting the Z direction and vice versa where the Z-position seems to be modifying the Y direction.

[image]

neophyl, 2020-12-03:
Have you previously  rotated the part that the modifier is applied to at all ?

Nanodiver38, 2020-12-03:
Yes, I used the quick place tool to put what would have been a plane normal
to the y-axis on the part to rotate it parallel to the z-axis. I figured
this wasn’t a real issue. Just thought I’d try and contribute.

Nanodiver38, 2020-12-03:
It’s funny because I didn’t rotate the cylinders though

neophyl, 2020-12-03:
The modifiers are tied to the parts though so when you rotate the part it’s has non obvious effects on manipulating the modifiers. In effect they aren’t referenced to what I would call the global frame but to the parts frame of reference.

Nanodiver38, 2020-12-03:
Ok. Shall I delete this thread?

bubnikv, 2020-12-04:
> shall I delete this thread?

Yes please.

closed 2020-12-08 by bubnikv