prusa3d/PrusaSlicer #5370: Slicer crashing with a flat orientation of an .stl when attempting to slice

CaptainCodswallop, 2020-12-03:
[Ender3LCDBoxBottom.zip](https://github.com/prusa3d/PrusaSlicer/files/5639152/Ender3LCDBoxBottom.zip)

### Version
2.3.0-alpha4

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Prusa Mini 4.2.1 (latest)

### Behavior
When the object included with the zip file is oriented flat on the bed in the most logical position for slicing it hangs at 30% for a few seconds and then slicer closes. No error messages shown. It works fine in the 2.2 release you can download.

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs
See the Zip File

hejllukas, 2020-12-03:
Could you try it if it also crashes in 2.3.0-beta1, please? I reproduce this issue in 2.3.0-alpha4, and it seems to be related to #5208 . This issue was fixed in 2.3.0-beta1.

CaptainCodswallop, 2020-12-03:
Will do. I will update later.

CaptainCodswallop, 2020-12-03:
Can confirm that it works perfectly with the Alpha 3 release. If you need anything else checked let me know.

bubnikv, 2020-12-04:
We would appreciate, if you could test with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta1 please

CaptainCodswallop, 2020-12-04:
Will do.

CaptainCodswallop, 2020-12-09:
Hi, I am sorry but I got side tracked. I did try this in the new Beta 2 release and can confirm that it sliced it no issues after the re orientation on the bed. I guess this can be marked as resolved? Thanks!

closed 2020-12-09 by CaptainCodswallop