prusa3d/PrusaSlicer #5373: Linux, GTK GUI window and elements not rendered correctly

fiechr, 2020-12-04:
### Version
2.2.0 (installed from Arch "community" repository (not AUR))

### Operating system type + version
Manjaro Linux (Arch based)
KDE/Plasma
gtk2 2.24.32-2
gtk3 1:3.24.23-4

```
PrusaSlicer
Version:   2.2.0
Build:     PrusaSlicer-2.2.0+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.9.11-3-MANJARO x86_64
Total RAM size [MB]: 16,748MB
OpenGL installation
GL version:   4.5 (Compatibility Profile) Mesa 20.2.3
Vendor:       X.Org
Renderer:     AMD PITCAIRN (DRM 2.50.0, 5.9.11-3-MANJARO, LLVM 11.0.0)
GLSL version: 4.50
Installed extensions:

GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_ballot
GL_ARB_shader_bit_encoding
GL_ARB_shader_clock
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shader_viewport_layer_array
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_meminfo
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_EGL_image_storage
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_depth_bounds_test
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_memory_object
GL_EXT_memory_object_fd
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_image_load_formatted
GL_EXT_shader_image_load_store
GL_EXT_shader_integer_mix
GL_EXT_shader_samples_identical
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shadow_lod
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_EXT_window_rectangles
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NVX_gpu_memory_info
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

### Behavior
Since a few months (possibility after a system update), default GUI rendering has become "ugly" again or seems to fall back to default non-decorated style.

I'm not sure if this is a build or packaging issue, but I remember that also happend with an older version, so this seems to be an issue or somehow a problem.

My other GTK applications look fine and I've set Plasma/KDE to use the default GTK look and feel.

bubnikv, 2020-12-04:
Please test with
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta1
We also have a specific GTK3 build.

fiechr, 2020-12-05:
Hi!

I've built with these options:

```
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_INSTALL_LIBDIR=lib \
    -DSLIC3R_FHS=ON \
    -DSLIC3R_PCH=OFF \
    -DSLIC3R_WX_STABLE=ON \
    -DSLIC3R_GTK=3 \
    -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-gtk3
```

Now the GUI looks better, but I get a lot of black blocks:

[image]
[image]

fiechr, 2020-12-06:
I also tried to run the binary release labeled "GTK3" in the archive filename, but I only get this error message:
[image]

I'm only using en_US and de_DE on this system. Maybe this is from an older configuration file? I guess, I could simply add en_GB locale, but I surely there must be a better solution. But that's another issue...

The actual issue seems to be related to WxWidgets 3.0 vs. 3.1, where the latter one doesn't seem to be available on Arch(?). Still, I'm wondering why the GUI was fine (without black bar/blocks) a couple of months ago.

bubnikv, 2020-12-06:
@fiechr 
The beta2 provides English US and English UK choices, see the beta2 change log.

bubnikv, 2020-12-06:
> This looks good so far (don't now about the drop-down menus, other drop downs in the PrusaSlicer GUI seem to be fine...)! I guess this is because WxWidgets is built in statically?

It is because supporting GTK on Linux with all the different distros, desktops, desktop skins and flavors is a nightmare.
Maybe @YuSanka will help.

fiechr, 2020-12-06:
> @fiechr
> The beta2 provides English US and English UK choices, see the beta2 change log.
> If you go to the language switching dialog, just select English US.

I have read it now. :) 
The problems was, that I couldn't continue after closing the error message, the application was closed, too. I've tried it multiple times. Only adding the en_GB system locale worked. In this case I would expect it to fall back to an existing locale?

bubnikv, 2020-12-12:
Please try with PrusaSlicer 2.3.0-beta3, @lukasmatena reworked the Linux specific code to align the locale with those installed on your Linux box.

fiechr, 2020-12-12:
I've removed the en_GB locale and started PrusaSlicer beta3 after that. Since there was no error message anymore, it seems to work now automatically with the installed en_US locale. I've not changed or reset any language settings besides removing the locale.

bubnikv, 2021-09-22:
> EDIT2: I've built and installed WxWidgets 3.1.4 now (using an already available AUR package) replacing the 3.0.x packages that were installed. The build now works with -DSLIC3R_WX_STABLE=OFF.
The result is the same as with the AppImage, all is fine, only the font in the selection drop down menus on the main screen is shifted downwards.

Please test with PrusaSlicer 2.4.0-alpha1.

closed 2023-02-06 by lukasmatena