prusa3d/PrusaSlicer #5376: Crash on generating gcode

Kohwalter80, 2020-12-04:
### Version
2.3.0-alpha4+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S (firmware is irrelevant)

### Behavior
* When I try to slice the appended file, it crashes, closing the program.
* I created on tinkercad the objects, exported as stl, drag and droped the stl file on the slicer, divided it into objects, automatic arranged them and click slice.

[relogio1.zip](https://github.com/prusa3d/PrusaSlicer/files/5645011/relogio1.zip)

#### Project File (.3MF) where problem occurs
Attached is the zipped 3MF file that is causing the issue.

bubnikv, 2020-12-04:
Most likely fixed in beta1.
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta1

closed 2020-12-04 by bubnikv

Kohwalter80, 2020-12-04:
Yes. With beta version it is ok.