prusa3d/PrusaSlicer #5377: Wipe Tower - IDEX - Tool change retracts and unretracts on same tool

soneshp, 2020-12-05:
### Version
2.2.0 up to 2.3.0 beta

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
IDEX - SKR Pro - Marlin 2

### Behavior
With relative E distances enabled, using a wipe tower, the active tool is retracting before the tool change, moves to wipe tower, and unretracts. THEN changes tool:

```
G1 X93.069 Y119.888 E0.38905 ; infill
**G1 E-50.00000 F2400.00000 ; retract for toolchange
G1 X96.734 Y85.147 F6000.000 ; Travel to a Wipe Tower
G1 E50.10000 F2400.00000 ; unretract**
;HEIGHT:0.100000
;TYPE:Wipe tower
;--------------------
; CP TOOLCHANGE START
; toolchange #3
; material : PLA -> PLA
;--------------------
M220 B
M220 S100
; CP TOOLCHANGE UNLOAD
G1  X96.234 Y84.897  
M900 K0
G1 F2000
G1  Y84.397  F2400
G4 S0
; Filament-specific end gcode 
;END gcode for filament

**T1 ; change extruder**

G1  X96.234 Y84.397  
; Filament gcode

G4 S0
G1  Y84.897 
; CP TOOLCHANGE WIPE
G1  X77.234  E0.3780 F1600
```
------------------------------------------------------------------------------------------
Disabling Wipe tower causes expected behavior:
```
G1 X87.649 Y111.231 E0.02201 ; infill
**G1 E-50.00000 F2400.00000 ; retract for toolchange**
; Filament-specific end gcode 
;END gcode for filament
**T1 ; change extruder**
; Filament gcode
G1 X85.572 Y108.116 F6000.000 ; move to first perimeter point
**G1 E50.10000 F2400.00000 ; unretract**
;TYPE:Perimeter
G1 F1200.000
G1 X85.846 Y107.881 E0.00644 ; perimeter
```

bubnikv, 2020-12-08:
The deretraction of the active tool head is a remnant of the single-extruder-multi-material origin of the wipe tower implementation.
@lukasmatena Please help with that, thanks.

Hades32, 2020-12-30:
Just found this, too, in the 2.3 RC2:
```
	Line 12063: T1 ; change extruder
	Line 12107: G1 E-2.00000 F2400.000 ; retract
	Line 12109: G1 E2.00000 F2400.000 ; unretract
	Line 12156: G1 E-2.00000 F2400.000 ; retract
	Line 12158: G1 E2.00000 F2400.000 ; unretract
	Line 12288: G1 E-100.00000 F2400.000 ; retract for toolchange
	Line 12292: G1 E100.00000 F2400.000 ; unretract
	Line 12357: T0 ; change extruder
```
Is there any workaround? 

BTW, why is the wipe tower not supported without relative addressing?

Hades32, 2020-12-30:
Working around with a simple regex:
[image]

BUT, you must also disable "prime all printing extruders" as there is NO RETRACTION on extruder change...

Hades32, 2020-12-30:
Another update: Also when printing without a wipe tower (but still with multimaterial single extruder setup), the slicer seems to ignore the initial "parking position" of the materials. I.e. there is NO unretraction when switching to a material for the first time (although it's of course in the parking position from the last print) but for subsequent toolchanges the unretraction is added correctly.

Should I open a new issue for this?

giPeteR, 2021-02-03:
One solution that works quite well is to add a few lines of Custom G-code in the Filament settings.
I have a dual extruder with Marlin. The script is only active when using the Wipe Tower.
You can strip all comments, they just help me find the sections in the G-code. I have plenty of custom scripts.

Filament Start G-code:
> ; -------------------------------------->>>
; Custom G-code: Filament  ABS - Start - START
; T[current_extruder]   ; Current extruder
{if wipe_tower}
  ; Wipe Tower used. Do extra de-retract.
  G1 E{retract_length_toolchange[current_extruder] + retract_restart_extra_toolchange[current_extruder]} F{60 * deretract_speed[current_extruder]} ; retract for toolchange
{endif}
; Custom G-code: Filament  ABS - Start - END
; <<<--------------------------------------

Filament End G-code:
>; -------------------------------------->>>
; Custom G-code: Filament  ABS - End - START
; T[current_extruder]   ; Current extruder
{if wipe_tower}
  ; Wipe Tower used. Do extra retract.
  G1 E-{retract_length_toolchange[current_extruder]} F{60 * retract_speed[current_extruder]} ; retract for toolchange
{endif}
; Custom G-code: Filament  ABS - End - END
; <<<--------------------------------------

I also have a post processing python script that adds an extra purge the first time the _second_ tool is activated. That's because initially I couldn't get the wipe tower to work. So if a print started with one tool the inactive tool was retracted a lot to prevent dripping, but then needed extra purge when starting.
Purge in front of the table didn't work since I have clips there to hold the bed.

klack, 2021-02-27:
Hello, I have created an easy to use post-processing script as a work around for this issue.  You can find it here:
https://gist.github.com/klack/915c5d86b6202fcb08af396ad3014fc9

lukasmatena, 2023-02-01:
This should be fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Feedback appreciated. Closing.

closed 2023-02-01 by lukasmatena