prusa3d/PrusaSlicer #5378: Switch the gcode flavor in slicer to RepRap, it will disable machine limits.

violetbp, 2020-12-05:
Setting gcode to RRF hides machine limits but still seems to use them.
increasing machine max acceleration, max feedrate, and max jerk in the INI file my print time dropped.

Switch the gcode flavor in slicer to RepRap, it will disable machine limits.
Related to #1089, #1234, #3776.
_Originally posted by @rtyr in https://github.com/prusa3d/PrusaSlicer/issues/4107#issuecomment-617197119_

rtyr, 2020-12-05:
It is now possible to disable machine limits without switching the gcode flavor (PrusaSlicer 2.3.0-beta).

[image]

rtyr, 2020-12-05:
Machine limits should not be used for time estimate with reprap/sprinter though.

lukasmatena, 2020-12-18:
This should be fixed (9605e23) in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Could you please retest and close the issue if happy? Thanks.

closed 2020-12-18 by lukasmatena