prusa3d/PrusaSlicer #5380: kossel linear plus: missing part of end gcode to turn off the bed

picard12, 2020-12-05:
### Version
_2.30.beta1 linux64bit

### Operating system type + version
Linux, kernel 2.5.7 sabayon 64bit

### 3D printer brand / version + firmware version (if known)
Kossel linear plus marlin 2.5.7.2

### Behavior
missing a final part of the end gcode that turns off the plate

_Is this a new feature request?_

I fix this issue adding this lines
M104 S0      ;extruder heater off
M140 S0      ;heated bed heater off
M107      ;fan off

rtyr, 2020-12-05:
What do you mean? I can see it there.

[image]

picard12, 2020-12-07:
I reset the prusaslicer setting by deleting the config folder and the run the autoconfig and the right parameter are like you are reporting.Ok.

closed 2020-12-07 by picard12