prusa3d/PrusaSlicer #5383: When brim is on, skirt interferes in the brim (2.3.0 alpha and beta)

premek81, 2020-12-05:
### Version 
2.3.0-alpha3 and 2.3.0-beta2

### Operating system type + version
MacOS 10.13.6

### 3D printer brand / version + firmware version (if known)
Original Prusa Mk3, stock

### Behavior
In older versions of PrusaSlicer (last tested in 2.1.1) when brim was off, skirt was close to the object as defined. When brim was on, skirt was around the brim.
In 2.3.0 (alpha and beta) the skirt stays in the same position whether brim is on or off, you need to go into menu and change skirt separation from objects
[image]
[image]

rtyr, 2020-12-05:
Intended behavior.

Why:
#724

Duplicate issue + links to many other similar issues:
#3692

Closing.

closed 2020-12-05 by rtyr

RajHid, 2020-12-26:
The brim also didn't touch the first object layer.
or My Layer height is poorly calibrated.
[image]
[image]
[image]

rtyr, 2020-12-26:
It takes elephant foot compensation into account, so it is expected. The gap shouldn't be present in the actuall print if your first layer/elephant foot compensation is set right.

First layer on your photo is not squished enough, you can see the gaps between individual brim lines.