prusa3d/PrusaSlicer #5384: [Feature Request] Profile comparaison

JulienThomas49, 2020-12-06:
### Version
_2.3.0 Beta 2_

### Operating system type + version
Operating System : **Windows**
System Architecture : **64 bit**
Windows Version : **Windows 10 (build 19041)**

### 3D printer brand / version + firmware version
Model : **Ender 3 V1**
Firmware version : **2.0.7.2**

### Feature request description
Add a comparison window which allows :
- Easy comparison between two profiles (even with an external profile from hard disk)
- Easy transfer values from one profile to another
- any other usefull comparison tools.

ShanDestromp, 2020-12-08:
To add to Juliens request (or provide my 'yes please' reasoning I guess), allow us to compare our profiles to the system profiles (particularly when the system profiles update).  Like many I have a series of modified system profiles saved because that's really the only way we can have persistent changes to the built in profiles.  Often those 'changes' are minor (eg I don't like the default infill pattern, or maybe I have fewer / more top & bottom layers but nothing else changed from default).

Whenever the system profiles are updated, the only (easy) way to compare is to save a new copy of the system profile and load both it and my version into a text editor and run a diff on them to see if there's anything worth changing, and then repeating this process for *every single profile I've modified*.

Obviously the mantra of 'if it's not broke don't fix it' applies to profiles, but then I (potentially) lose out on new features (ex infill anchors) and overall refinements.  The current / new window that pops up when changing profiles is a good step in this direction but (AFAICT) only activates when you have *unsaved changes* in your profile, and only shows those unsaved changes.

### My specific proposal:  
Allow this dialog to be manually called with any two profiles and compare / transfer *all* differences between them.

Kachidoki2807, 2020-12-21:
My two cents, IdeaMaker already propose a limited version (view only) of this comparison tool. It is a useful feature to instantly get the differences between updated profiles, but even more between different profiles, ie: between quality and speed profiles, or even between profiles from different printers.

The current inheritance system already help for that, thanks to the lock/unlock icon, but it does means the observed profile must inherit from a reference profile. It is a very limited but already powerful feature.

I am writing a vendor profile for an i3 based IDEX printer with a dozen of profiles, which inherit for the most part from the mk3s settings as the architecture and extruders are very similar. But now I have no way to easily compare my settings into PrusaSlicer, nor at the building of the profile since the .ini files are completely different. I have to export them one by one afterward to have a side-by-side comparable file.

The proposal of @ShanDestromp is a easy first step into this direction, and can even be enough to cover the most of use cases.

LightningShark25, 2021-02-25:
+1 
I can see the merits of the inheritance feature HOWEVER, I don't necessarily need inheritance.  I'm likely to stuff things up by making it this complicated.   I can see myself accidentally updating a setting in one profile, forgetting the inheritance, then not realising its changed in another profile.

rtyr, 2021-02-25:
Same as https://github.com/prusa3d/PrusaSlicer/issues/1566.

Ic33s, 2022-08-14:
As my request is closed. I want to add here, that at least for some settings (e.g. gcode for start and end) it would make sense to be able to freely edit them in the compare/merge view than, instead of only "copy left" and "copy right".

kubispe1, 2022-12-15:
> Easy comparison between two profiles (even with an external profile from hard disk)
Easy transfer values from one profile to another
any other usefull comparison tools.

This FR will be implemented with upcoming release 2.6.0. It is build into Compare Presets
Thanks

closed 2022-12-15 by kubispe1