prusa3d/PrusaSlicer #5385: Avoid crossing perimeters does nothing?

ochm, 2020-12-06:
### Version
2.3.0-beat2+win64

### Operating system type + version
W10

### 3D printer brand / version + firmware version (if known)
Doesn't mater

### Behavior
I tried fews models (also two or more models on bed) with avoid xing perimeters on or off, and visually i see no changes in travels. Does this function work or not?

fiechr, 2020-12-06:
Have you tried to compare the generated G-Code output using a diff tool?

ochm, 2020-12-06:
> 
> 
> Have you tried to compare the generated G-Code output using a diff tool?

Not, i believe slicer. If there is visualisation, why to debug gcode?

hejllukas, 2020-12-06:
Hi, yes, this feature in general works. The way it works is not optimal, and some travels cross perimeters even if they don't have to. Also, there is a couple of reported issues for this feature. But for most models, should some travels be different. So, if you have some models for which this feature does nothing, please post 3MF files. 

We are also working on revamping this feature and should work better than the current implementation and also fixed many reported issues. But this version is not present in 2.3.0-beta2. In 2.3.0-beta2, this feature works in the same way as in 2.2.0.

In short, this feature tries to keeps travels inside an object. If travel leads between two objects, then obviously, perimeters are crossed for both these objects but should avoid other objects. This is the same if the printed layer of one model contains more separated parts (islands).

ochm, 2020-12-07:
Hi Lukas, you are right, i tested function on two objects, but three objects in line are better.
[image]
Closing this issue.

closed 2020-12-07 by ochm