prusa3d/PrusaSlicer #5386: Various Build issues

frispete, 2020-12-06:
### 2.2.0 and 2.3.0~0.20201205T120038.ac0584958
### openSUSE 15.2 and Tumbleweed
### Prusa Mini+ (if delivered one day..)
### Build issues

Hi, I'm in the process for preparation of our 3D print environment for my son and me. The Mini+ is ordered, but delivery seems to be delayed far beyond Christmas. Meanwhile, I set up the tool chain for our openSUSE Tumbleweed boxes. My son is rather fluent in Blender, while I'm somewhat experienced in packaging. For 3D Printing, the packages are build in public [here](https://build.opensuse.org/project/monitor/home:frispete:3dprinting). In terms of PrusaSlicer, I started with packaging [2.2.0](https://build.opensuse.org/package/show/home:frispete:3dprinting/PrusaSlicer) and [2.3.0](https://build.opensuse.org/package/show/home:frispete:3dprinting/PrusaSlicer-git): ac0584958260ea2bc9ad0efc04d289c3baee9ed1.

Tumbleweed features a pretty current tool chain of everything (gcc10, cmake 3.19.1, ...). 

* Enabling ASAN results in build failures
* The git build (version_2.3.0-beta2) insists of binding to a static libz (shared devel lib is not properly located by cmake (might be a cmake 3.19.1 issue, though)
* Leap 15.2: wxstable build fails with:
```
[  355s] cd "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r" && /usr/bin/c++  -DBOOST_LOG_DYN_LINK -DLIBNEST2D_GEOMETRIES_clipper -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxDEBUG_LEVEL=0 -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/platform" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/clipper" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/polypartition" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/hidapi/include" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/libslic3r" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/libnest2d/include" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/miniz" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/glu-libtess/include" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/qhull/src" -isystem /usr/include/eigen3 -isystem /usr/lib64/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -isystem "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/libigl" -isystem /usr/include/OpenEXR  -fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -pthread -g -fPIC   -DDEBUG -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -pthread -std=gnu++1z -Winvalid-pch -include "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx" -o CMakeFiles/libslic3r_gui.dir/GUI/UnsavedChangesDialog.cpp.o -c "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r/GUI/UnsavedChangesDialog.cpp"
[  355s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r/GUI/UnsavedChangesDialog.cpp: In member function 'void Slic3r::GUI::ModelNode::UpdateIcons()':
[  355s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r/GUI/UnsavedChangesDialog.cpp:196:47: error: operands to ?: have different types 'wxString' and 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}'
[  355s]          m_old_color_bmp = get_bitmap(m_toggle ? m_old_color : grey);
[  355s]                                       ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
[  355s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r/GUI/UnsavedChangesDialog.cpp:196:47: note:   and each type can be converted to the other
[  355s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r/GUI/UnsavedChangesDialog.cpp:198:47: error: operands to ?: have different types 'wxString' and 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}'
[  355s]          m_new_color_bmp = get_bitmap(m_toggle ? m_new_color : grey);
[  355s]                                       ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
[  355s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r/GUI/UnsavedChangesDialog.cpp:198:47: note:   and each type can be converted to the other
```

The Leap builds aren't that important for now, but the former issues would be nice to resolve..

frispete, 2020-12-06:
Okay, second issue was triggered from a library relocation issue. Please ignore.

bubnikv, 2020-12-07:
Now I am not sure whether I closed the issue rightly. Is there any issue left?

frispete, 2020-12-07:
Hi Vojtěch, 

the ASAN issue troubles me a bit. Will do further builds in this respect and report.

Also, I would prefer to use the system provided qhull library, if possible. 
I can dive into the related cmake scaffolds, but I'm not that fluent in cmake country..

The wxstable incompatibility is worth to be solved, isn't it? It's merely missing a cast, and might be evoked by pretty strict compiler settings, but it's always preferable to fix such issues in code instead of loosening the compiler checks. Don't you think so as well?

All in all, it's worth to reopen this issue. At least, from my POV.

frispete, 2020-12-07:
Here's the ASAN issue, it fails in the encoding-check:
```
[    7s] Scanning dependencies of target encoding-check
[    7s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] make  -f src/build-utils/CMakeFiles/encoding-check.dir/build.make src/build-utils/CMakeFiles/encoding-check.dir/build
[    7s] make[2]: Entering directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] [  2%] Building CXX object src/build-utils/CMakeFiles/encoding-check.dir/encoding-check.cpp.o
[    7s] cd "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/build-utils" && /usr/bin/c++ -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/platform" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/clipper" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/polypartition" -isystem /usr/include/eigen3 -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -g -fPIE -DDEBUG -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -fsanitize=address -fno-omit-frame-pointer -std=gnu++17 -o CMakeFiles/encoding-check.dir/encoding-check.cpp.o -c "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/build-utils/encoding-check.cpp"
[    7s] [  2%] Linking CXX executable encoding-check
[    7s] cd "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/build-utils" && /usr/bin/cmake -E cmake_link_script CMakeFiles/encoding-check.dir/link.txt --verbose=1
[    7s] /usr/bin/c++ -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -g -flto=auto -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now -fsanitize=address -lasan CMakeFiles/encoding-check.dir/encoding-check.cpp.o -o encoding-check 
[    7s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] [  2%] Built target encoding-check
[    7s] make  -f src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/build.make src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/depend
[    7s] make[2]: Entering directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] cd "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build" && /usr/bin/cmake -E cmake_depends "Unix Makefiles" "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958" "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r" "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build" "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r" "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/DependInfo.cmake"
[    7s] Dependee "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/DependInfo.cmake" is newer than depender "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/depend.internal".
[    7s] Dependee "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/depend.internal".
[    7s] Scanning dependencies of target encoding-check-libslic3r_gui
[    7s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] make  -f src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/build.make src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/build
[    7s] make[2]: Entering directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] [  2%] Checking source files encodings for target libslic3r_gui
[    7s] cd "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/src/slic3r" && "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build/src/build-utils/encoding-check" libslic3r_gui pchheader.cpp
[    7s] ==23488==ERROR: AddressSanitizer failed to allocate 0xdfff0001000 (15392894357504) bytes at address 2008fff7000 (errno: 12)
[    7s] ==23488==ReserveShadowMemoryRange failed while trying to map 0xdfff0001000 bytes. Perhaps you're using ulimit -v
[    7s] make[2]: *** [src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/build.make:291: encoding-check-libslic3r_gui] Aborted (core dumped)
[    7s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] make[1]: *** [CMakeFiles/Makefile2:1440: src/slic3r/CMakeFiles/encoding-check-libslic3r_gui.dir/all] Error 2
[    7s] make[1]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.3.0~0.20201205T120038.ac0584958/build'
[    7s] make: *** [Makefile:163: all] Error 2
[    7s] error: Bad exit status from /var/tmp/rpm-tmp.tB7j7f (%build)
```

l29ah, 2021-01-12:
I have a problem with prusaslicer's cmake locating the system qhull on Gentoo as well.

frispete, 2021-12-11:
Let's close this one, sorry for not following up in time.. Those issues were long fixed after all.

Happily using the Mini+ for almost a year now.

closed 2021-12-11 by frispete