prusa3d/PrusaSlicer #5387: MAC IOS - After slicing the object disappears

RudrigoMaciel, 2020-12-06:
### Version
PrusaSlicer-2.3.0-beta2+-202012051103

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS High Sierra Version 10.13.6 (Build 17G14033)
Total RAM size [MB]: 4,295MB
OpenGL installation
GL version: 2.1 INTEL-10.4.14
Vendor: Intel Inc.
Renderer: Intel HD Graphics 3000 OpenGL Engine
GLSL version: 1.20

### 3D printer brand / version + firmware version (if known)
PrusaSlicer issue, not printer.

### Behavior
* _Describe the problem_
After slicing, the object disappears from the plate for viewing.
Obs: During slicing, just berfore the final slicing step (message "generating code"), the object is on the plate and can be previewed for the last processing, but right after the message "slicing complete", the object disappears from the visualization.

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

1) During final slicing process (generate code (last preview is on the plate):

[image]

2) After slicing:
[image]
[image]
[image]

_Is this a new feature request?_
Yes, since PrusaSlicer-2.3.0 alpha versions

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[belle_comSuporte_v5.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5648920/belle_comSuporte_v5.3mf.zip)

bubnikv, 2020-12-07:
>    1. After slicing:

Total RAM size [MB]: 4,295MB

You may simply be low on RAM.

FidelCapo, 2020-12-07:
I cannot reproduce this issue on mac. But I noticed a differences between settings on screenshots and settings in 3mf file like Brim or Infill density..

RudrigoMaciel, 2020-12-07:
> […](#)

Ok. I used this old mac for all versions of PrusaSlicer without issues until the version PrusaSlicer-2.3.0. Does this version require more RAM? I don't have another MAC to do this test locally where I am now.

RudrigoMaciel, 2020-12-07:
> I cannot reproduce this issue on mac. But I noticed a differences between settings on screenshots and settings in 3mf file like Brim or Infill density..

I attached the generated 3mf file PrusaSlicer-2.3.0-beta2 + -202012051103, but made the screenshots using the same object, but with the 3mf generated by PrusaSlicer-2.2.0 + -202003211132. The issue is the same, no matter which 3mf you use. It was a test. Attached, the 3mf generated by PrusaSlicer-2.2.0 + -202003211132.

ado366, 2021-01-21:
I seem to have the same problem. 
The final preview which I am assuming is new, does not seem to be supported on older Macs. It would be great to be able to disable this new preview for older maschines

enricoturri1966, 2021-01-21:
Could you please test if the following experimental build helps in fixing this issue ?

https://www.dropbox.com/s/wgivr3k8vwyqebc/PrusaSlicer-2.3.0%2B64-et-splitted-vbuffer-g0f8e7f3e8-202101210845.dmg?dl=0

ado366, 2021-01-21:
> Could you please test if the following experimental build helps in fixing this issue?
> 
> https://www.dropbox.com/s/wgivr3k8vwyqebc/PrusaSlicer-2.3.0%2B64-et-splitted-vbuffer-g0f8e7f3e8-202101210845.dmg?dl=0

I just did a quick test. Seems to be working perfectly for me. ^_^

bubnikv, 2021-01-21:
@RudrigoMaciel  you mean that the version from dropbox is working fine for you, while the official PrusaSlicer 2.3.0 release does not?

RudrigoMaciel, 2021-01-21:
> version from dropbox is working fine for you, while the official PrusaSlicer 2.3.0 release

Sorry. The official PrusaSlicer 2.3.0 release is working fine for you. I didn't test the dropbox version.

RudrigoMaciel, 2021-01-21:
> > version from dropbox is working fine for you, while the official PrusaSlicer 2.3.0 release
> 
> Sorry. The official PrusaSlicer 2.3.0 release is working fine for you. I didn't test the dropbox version.

I do a teste with the dropbox version (Versão 2.3.0+64-et-splitted-vbuffer), and it's works nice too.

bubnikv, 2021-09-22:
Fixed, closing.

closed 2021-09-22 by bubnikv