prusa3d/PrusaSlicer #5389: Ironing feature leads to clogged print nozzle.

codemonkey2k5, 2020-12-06:
### Version
2.3.0 Beta 2

### Operating system type + version
Windows 10 Pro 2009

### 3D printer brand / version + firmware version (if known)
Prusa MK3 Firmware 3.9.1

### Behavior
I've been playing with the latest beta for a day or two, really like it.  
But today I had a large print that I wanted to use the new iron feature on.  
This print had about half of the print bed to iron.   

I used the default settings by selecting Ironing and all top surfaces.  

I believe what is happening is that the PLA is at temperature for so long and really not extruding all that much during the process.  The result was burnt PLA clogging up my print head.  

Was printing Hatchbox Silver PLA at 210C on to a 60C bed.   Never have issues with these settings.   
This feature may need some tweaking or better documentation to avoid this issue.

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
Here is the print job:
[First_pandemic_2.zip](https://github.com/prusa3d/PrusaSlicer/files/5649771/First_pandemic_2.zip)

codemonkey2k5, 2020-12-06:
Hey guys, I zipped up the wrong .3MF    Only difference is that did this print with Ironing on ALL TOP Surfaces.  This zip is only the very top of the print.  It's likely that the same thing would happen though.

bubnikv, 2020-12-07:
> I believe what is happening is that the PLA is at temperature for so long and really not extruding all that much during the process.  The result was burnt PLA clogging up my print head.  

The issue is a heat creep, see the "downsides" section
https://help.prusa3d.com/en/article/ironing_177488

codemonkey2k5, 2020-12-07:
Thank you, I didn't know about that knowledgebase.  Super helpful.

closed 2020-12-08 by bubnikv

fe60, 2020-12-08:
I faced the same problem while my Prusa i3 MK3S tried to iron a couple of identical parts.
Just adding some facts/numbers:
Material: PLA
Ironing Settings: default (Flow rate: 15%, spacing between passes: 0.1mm)
Surface until nozzle got clogged: 2x 9.5cm<sup>2</sup>

Avamander, 2025-07-05:
@bubnikv 
Would it be possible to introduce an option to print one of those extruder cleaning blobs on the build plate edge after a certain amount `n` mm^2 of ironing done within `m` layers? I think people have shot themselves in the foot themselves quite a few times when ironing causes (even slight) clogs. This could help.