prusa3d/PrusaSlicer #5390: Ironing probably not excluding tiny surfaces

edolis, 2020-12-07:
### Version
2.3.0.beta 1

I think that ironing does not exclude very small surfaces like embossed letters - this is destroying them.
Also, does not accept 0% flow rate- i think it should, if we are talking of pure ironing...

Left: with ironing, 1% flow rate 
Right: no ironing

[image]

fe60, 2020-12-08:
Just in case you are not aware of this: As a workaround you can block ironing using infill modifiers https://github.com/prusa3d/PrusaSlicer/issues/5116#issuecomment-740081975.

sglux, 2022-05-23:
Also relates to minimum ironing area exclusion #6928
May be closed in favour of that issue?

stepikovo, 2022-05-29:
I'll leave this open because of the feature request for a 0 % flow rate, I just changed the title to be unique

closed 2026-09-29 by prusaslicer-bot[bot]