prusa3d/PrusaSlicer #5391: Prusa slicer gcode only prints one line then no movement on extruder anymore

bowie130459, 2020-12-07:
### Version
Prusa Slicer 2.2.0+

### Operating system type + version
Mac OSX 10.15.7

### 3D printer brand / version + firmware version (if known)
Ender 3d Pro with TH3D_UFW_U1.R2.A8 and Tintan Aero 3d direct extruder version

### Behavior
Hallo all,

The gcode on the Prusasliced modell looks just normal. here are the first entries: It is a direkt Fileament extruder.

; generated by PrusaSlicer 2.2.0+ on 2020-12-07 at 09:27:02 UTC

;

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M201 X500 Y500 Z100 E5000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z10 E60 ; sets maximum feedrates, mm/sec
M204 P500 R1000 T500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S215 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp
G28 ; home all
G1 Z2 F240
G1 X2 Y10 F3000
G1 Z0.28 F240
G92 E0.0
G1 Y190 E15.0 F1500.0 ; intro line
G1 X2.3 F5000
G1 Y10 E30 F1200.0 ; intro line
G92 E0.0
M420 S1
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
;BEFORE_LAYER_CHANGE
;0.2

G1 E-5.00000 F3600.00000
G92 E0
G1 Z0.200 F9000.000
;AFTER_LAYER_CHANGE
;0.2
G1 X97.085 Y98.856
G1 E5.00000 F2400.00000
G1 F1200.000
G1 X97.632 Y97.951 E5.03315
G1 X98.093 Y97.517 E5.05299
G1 X98.627 Y97.186 E5.07270
G1 X99.221 Y96.967 E5.09254
G1 X99.847 Y96.873 E5.11238
G1 X120.004 Y96.869 E5.74441
G1 X120.578 Y96.923 E5.76248
G1 X121.132 Y97.081 E5.78054
G1 X121.648 Y97.338 E5.79861
G1 X122.108 Y97.685 E5.81668
G1 X122.497 Y98.110 E5.83475

The Extruder movement values aren't zero. The above mentioned intro lines are printed as expected.

Put then while printing on the printer the interpreted Gcode loooks like this:

10:35:33: Printing layer 1 of 87
< 10:35:33: X:2.30 Y:10.00 Z:0.28 E:0.00 Count X:0 Y:0 Z:61
< 10:35:50: echo:Bed Leveling Ein
< 10:35:50: X:2.30 Y:10.00 Z:0.29 E:0.00 Count X:184 Y:800 Z:112
< 10:35:50: X:2.30 Y:10.00 Z:0.29 E:0.00 Count X:184 Y:800 Z:112
< 10:35:51: X:2.30 Y:10.00 Z:0.20 E:0.00 Count X:184 Y:800 Z:112
10:35:51: Printing layer 2 of 87
< 10:36:12: X:98.94 Y:97.88 Z:0.20 E:0.00 Count X:7994 Y:9790 Z:30
< 10:36:39: X:106.55 Y:105.64 Z:0.20 E:0.00 Count X:8462 Y:9044 Z:34
< 10:36:53: X:100.60 Y:100.31 Z:0.20 E:0.00 Count X:8019 Y:8017 Z:44
< 10:37:07: X:117.54 Y:101.62 Z:0.20 E:0.00 Count X:9343 Y:8254 Z:40
< 10:37:11: X:118.38 Y:109.62 Z:0.20 E:0.00 Count X:9470 Y:8959 Z:34
< 10:37:13: X:115.16 Y:114.15 Z:0.20 E:0.00 Count X:9294 Y:8994 Z:33
< 10:37:16: X:109.62 Y:118.38 Z:0.20 E:0.00 Count X:8600 Y:9470 Z:30
< 10:37:18: X:104.72 Y:118.38 Z:0.20 E:0.00 Count X:8468 Y:9469 Z:31
< 10:37:21: X:101.62 Y:115.66 Z:0.20 E:0.00 Count X:8223 Y:9374 Z:32
< 10:37:22: X:101.62 Y:111.44 Z:0.20 E:0.00 Count X:8345 Y:9253 Z:32
< 10:37:26: X:103.83 Y:107.57 Z:0.20 E:0.00 Count X:8276 Y:8726 Z:37
< 10:37:27: X:110.91 Y:101.62 Z:0.20 E:0.00 Count X:8642 Y:8130 Z:42
< 10:37:32: X:117.54 Y:118.38 Z:0.32 E:0.00 Count X:9344 Y:9470 Z:29
10:37:32: Printing layer 3 of 87
< 10:37:47: X:122.22 Y:120.81 Z:0.32 E:0.00 Count X:7994 Y:7810 Z:94

the Extruder movement values are zero and so no Filament is coming out of the nozzle.

The same Stl sliced with cura is Printing all layers as expected.

What is happenning here. The same Printer is used but different Slicers.

Could any one please help me.

 

Kind regards

Willi

_Is this a new feature request?_ Yes

#### Project File (.3MF) where problem occurs
[PLA-3DDC_paradroid_Testcube_V2_0.12mm_PLA_ENDER3_34m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5653220/PLA-3DDC_paradroid_Testcube_V2_0.12mm_PLA_ENDER3_34m.gcode.zip)

[PLA-3DDC_paradroid_Testcube_V2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5653211/PLA-3DDC_paradroid_Testcube_V2.3mf.zip)

rtyr, 2020-12-07:
Your gcode and 3MF seems fine. It looks like some limitation of the TH3D unified firmware. Our E values may be too precise for it for some reason (I believe it is based on Marlin fw). You should probably ask TH3DSTUDIO.

bowie130459, 2020-12-08:
Hello, this may be ok but I found the issue which caused the stop of the extruder motor after prining the initial lines.
It is the gcode section shown below with coments of my findings while testing.
 Filament gcode
;BEFORE_LAYER_CHANGE
;0.2

G1 E-5.00000 F3600.00000 --------------This is the error, this prevents the Extruder to work from now on 
G92 E0                         --------------------This is OK 
G1 Z0.200 F9000.000 --------------------This is OK
;AFTER_LAYER_CHANGE
;0.2

After deleting the marked as Error Line the Printer works as expected. 
I did not find the section within the Prusaslicer software to chaneg this.

So it could be a big thing If this could be solved. 
I just shown this also to TH3dStudio

Kind regards Willi

bubnikv, 2020-12-08:
Maybe PrusaSlicer 2.3.0-beta2 fixes your issue?
Beta2 fixes https://github.com/prusa3d/PrusaSlicer/issues/5362
which seems to be exactly your issue.

bowie130459, 2020-12-08:
Hi, seems to be correct that this is my issue.

then no movement on extruder anymore (#5391)
Datum: 2020-12-08T10:44:12+0100
 
 
 
 

"https://github.com/prusa3d/PrusaSlicer/issues/5391#issuecomment-740507838", 
"url": 
"https://github.com/prusa3d/PrusaSlicer/issues/5391#issuecomment-740507838", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

bubnikv, 2020-12-08:
So coud you retest with PrusaSlicer 2.3.0-beta2 and close this issue?

bowie130459, 2020-12-08:
I just tested it again and again: It's the same Problem.
All commands looking like the following are stopping the extruder for the rest of the file.
G1 E-5.00000 F3600.000
or aquivalent commands.
Kind regards
Willi

bubnikv, 2020-12-08:
I am sorry, but you have to resolve your issue with our printer provider.
Closing.

closed 2020-12-08 by bubnikv