prusa3d/PrusaSlicer #5392: Broken end gcode for ender 5 in ps2.3.0-beta2

definitely-not-a-t-rex, 2020-12-07:
### Version
_Version of PrusaSlicer used goes here_2.3.0-beta2

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_windows 10 20h2 (19042.630)
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) ender 5 with klipper 0.9.1
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ Unable to slice, slicing returns invalid data and error relative to end gcode from that profile 
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ a sliced gcode
* _Actual Results_ window with 
```
G-code export to C:\Users\ange9\AppData\Local\Temp\.15904.gcode failed due to invalid custom G-code sections:
end_gcode
Parsing error at line 3. Expecting tag literal-char
{if max_layer_z < max_print_height-10}G1 Z{z_offset+max_print_height-10)} F600{endif} ; Move print bed down further down
                                                                       ^
Please inspect the file C:\Users\ange9\AppData\Local\Temp\.15904.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.
```

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_ 
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-12-07:
Z{z_offset+max_print_height-10)}

The arrow points to a brace, which you should remove.

definitely-not-a-t-rex, 2020-12-07:
> 
> […](#)

I somehow totally missed it, thank you!

closed 2020-12-07 by definitely-not-a-t-rex