prusa3d/PrusaSlicer #5393: Import config bundle with physical printers

fiechr, 2020-12-07:
### Version
2.3.0 beta2

### Operating system type + version
Linux 5.9.11, Manjaro
Plasma/KDE

### Behavior

#### Describe the problem
I have exported a config bundle including the physical printers using the new menu option.
Now I removed all my presets and want to import the same file back in.
But the physical printers don't show up in the drop-down menu, only the Printer User Presets (without network data).

#### Expected Results
After import all physical printers should show up -- as they did at the time of export

#### Actual Results
Only Printer User Presets show up

fiechr, 2020-12-07:
It seems, I need to restart the program first, to get the physical printers.

bubnikv, 2020-12-12:
Fixed in PrusaSlicer 2.3.0-beta3

closed 2020-12-12 by bubnikv