prusa3d/PrusaSlicer #5394: nVidia Overlay causes Startup Crash

Orochium, 2020-12-07:
### Version
PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10 (build 18363), 64-bit edition

### Behavior

Issue observed: 

On startup, Prusa-slicer.exe will close unexpectedly without an error message. Windows event viewer indicated a HEAP corruption occured in NTDLL.DLL (observed in last 6 drivers released via GeForce Experience); specific dump information seems to indicate an issue with the NVidia in-game overlay due to how the graphics driver is called for hardware acceleration in Prusa-Slicer.  However, nVidia doesn't provide a method to disable overlay for specific applications.

To Reproduce:

With any nVidia GTX or RTX graphics card with most recent drivers:
- Launch Prusa-slicer.exe
- Application quits unexpectedly
- nVidia overlay banner appears

What was expected:
- Launch Prusa-slicer.exe
- Application operates normally
- nVidia overlay banner appears

This is not a feature request but a work around is presented below.

#### Current work-arounds

Disable NVidia In-game Overlay:

1. Right click on the nVidia system tray icon
2. Select "NVidia GeForce Experience"
3. Click the settings Cog in the application
4. Under the "General" tab, select and disable "In-Game Overlay" 
5. Re-open Prusa-Slicer and it should now launch with hardware acceleration

Disable Hardware acceleration in Prusa-slicer (Slow):

1. Navigation to C:\Program Files\Prusa3D\PrusaSlicer\mesa
2. Copy "opengl32.dll"
3. Paste "opengl32.dll" into C:\Program Files\Prusa3D\PrusaSlicer\
4. Re-open Prusa-Slicer and it should now launch with software rendering

This information is provided as-is for development purposes and bug-tracking. Please ask if you need additional information.

bubnikv, 2020-12-07:
Would you please test with
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta2 ?

Orochium, 2020-12-07:
Alright, just ran the 2.3.0-beta2 through a battery of tests.

The issue appears to be cleared up in 2.3.0-beta2; 

I tried loading it from a drop folder on my desktop, and then also manually putting it into "c:\Program Files\Prusa3D2\PrusaSlicer" to see if it was at all permissions related but it successfully loads in both locations. I took the same steps in 2.3.0-beta2 as I was to make the 2.2.0 version crash (2.2.9.1 according to the installer)

Interesting note:
If I load 2.3.0-beta2 on my desktop first, then load 2.2.0, it loads without crashing. If I try to load 2.2.0 without it, it goes back to crashing. 

Either way, looks resolved in the Beta

bubnikv, 2020-12-08:
Cool, thanks. 
 
> Interesting note:
> If I load 2.3.0-beta2 on my desktop first, then load 2.2.0, it loads without crashing. If I try to load 2.2.0 without it, it goes back to crashing.

The ways of vendor graphics drivers are mysterious.

closed 2020-12-08 by bubnikv