prusa3d/PrusaSlicer #5397: Bug: Unwanted filament changes triggered when using Hilbert Curve infill

JasonGuthrie, 2020-12-07:
### Version
2.3.0 Alpha3

### Operating system type + version
W10 - Pro

### 3D printer brand / version + firmware version (if known)
Prusa Mk3 + V3.9.0 (1.0.6)

### Behavior
When using Hilbert Curve infill, mysterious filament changes are requested by the printer. This causes no issue to print quality but requires the machine to be attended to for a needless filament change every 10 minutes or so.

The filament changes are not visible in the slicer preview or the g-code preview, they only show their faces when actually printing.

_Is this a new feature request?_
No

rtyr, 2020-12-07:
Please provide your 3MF project. 

I don't think it's PrusaSlicer related issue. It is most likely false filament runout detection. Does it happen with filament sensor disabled (can be disabled on lcd)?

JasonGuthrie, 2020-12-07:
> I don't think it's PrusaSlicer related issue. It is most likely false filament runout detection. Does it happen with filament sensor disabled (can be disabled on lcd)?

Ah good point, perhaps the infill's small but rapid accelerations are upsetting the filament sensor (although I am running black PETG so I thought it would have pretty robust optical properties). 

I will try again with sensor turned off I will and report back with the issue. 

I've attached the 3MF for additional reference. 

[Bottom with Hilbert Curve.zip](https://github.com/prusa3d/PrusaSlicer/files/5655820/Bottom.with.Hilbert.Curve.zip)

JasonGuthrie, 2020-12-08:
> I don't think it's PrusaSlicer related issue. It is most likely false filament runout detection. Does it happen with filament sensor disabled (can be disabled on lcd)?

It was indeed the filament sensor. I tried the print again with it turned off and the problem resolved itself. 

Thank you for the suggestion!

closed 2020-12-08 by JasonGuthrie