prusa3d/PrusaSlicer #5398: [2.3.0] Place on Face doesn't always work

ShanDestromp, 2020-12-07:
This may be related to #5172 and #5011 (?) but since my results were different I figured a separate BR was appropriate.

### Version
2.3.0A4, 2.3.0B1 and 2.3.0B2 Win64.  Bug is **NOT** present in 2.2.0 Win64

### Operating system type + version
Win10 64 19042

### 3D printer brand / version + firmware version (if known)
i3Mk3

### Behavior
* _Describe the problem_
Add attached STL to build platform, attempt to place on bottom face (red arrow screenshot)
Object is simply inverted but remains at the 15° angle.  Repeated attempts result in just going between original and inverted orientation.

Selecting *any other* (including the "top") face to lay down works as expected, and **from then on** you can "place" with the indicated face.

* _Steps needed to reproduce the problem_
See previous.
* _Expected Results_
To be able to "place on face" as expected.
* _Actual Results_
Not that.
* _Is this a new feature request?_
No

### Additional note:

#### Project File (.3MF) where problem occurs
[PlaceOnFaceError3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/5655605/PlaceOnFaceError.zip)

[image]

bubnikv, 2020-12-08:
https://github.com/prusa3d/PrusaSlicer/commit/feaf4192c7f8f4914698ace630a1032ddc00fa75 may fix it.
@lukasmatena review the fix.

lukasmatena, 2020-12-09:
Should be fixed with 2d6b694 for PrusaSlicer 2.3.0-beta3. Closing.

closed 2020-12-09 by lukasmatena