prusa3d/PrusaSlicer #5400: Tests fail with std::bad_alloc

hyperair, 2020-12-08:
### Version

2.2.0

### Operating system type + version

Debian unstable

### Behavior

Running tests after a fresh compilation of 2.2.0 from source fails with `std::bad_alloc`. See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=975198 for more information.

bubnikv, 2020-12-08:
Thanks for report, however we are putting all our effort into 2.3.0 release. We would appreciate your help with packaging 2.3.0

hyperair, 2020-12-27:
Just tried building 2.3-rc2. Failed with the same error.

bubnikv, 2021-01-05:
Is there any RAM limitation when running the test?

hyperair, 2021-01-07:
Ah, no there aren't. I've found the error, it turns out one of the libraries in Debian is buggy and exporting `Catch::` symbols, causing memory corruption while running tests -- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=978617

hyperair, 2021-01-07:
I've hacked around this issue using this patch for now. I haven't uploaded it yet though.

```
From: Chow Loong Jin <hyperair@debian.org>
Date: Tue, 29 Dec 2020 18:22:19 +0800
Subject: Move Catch namespace into PrusaCatch

This is a hack to work around Catch symbols showing up in liblog4cplus:
https://bugs.debian.org/978617
---
 tests/catch2/catch.hpp                   | 651 ++++++++++++++++---------------
 tests/catch2/catch_reporter_automake.hpp |   6 +-
 tests/catch2/catch_reporter_tap.hpp      |   5 +-
 tests/catch2/catch_reporter_teamcity.hpp |   6 +-
 tests/catch2/prusa-catch-alias.hpp       |   7 +
 tests/catch_main.hpp                     |   4 +-
 6 files changed, 346 insertions(+), 333 deletions(-)
 create mode 100644 tests/catch2/prusa-catch-alias.hpp

diff --git a/tests/catch2/catch.hpp b/tests/catch2/catch.hpp
index 282d156..345c0d9 100644
--- a/tests/catch2/catch.hpp
+++ b/tests/catch2/catch.hpp
@@ -1,3 +1,4 @@
+#include "prusa-catch-alias.hpp"
 /*
  *  Catch v2.13.3
  *  Generated: 2020-10-31 18:20:31.045274
@@ -92,7 +93,7 @@
 
 // start catch_user_interfaces.h
 
-namespace Catch {
+namespace PrusaCatch {
     unsigned int rngSeed();
 }
 
@@ -476,7 +477,7 @@ namespace Catch {
 struct Catch_global_namespace_dummy {};
 std::ostream& operator<<(std::ostream&, Catch_global_namespace_dummy);
 
-namespace Catch {
+namespace PrusaCatch {
 
     struct CaseSensitive { enum Choice {
         Yes,
@@ -519,7 +520,7 @@ namespace Catch {
 
     // Bring in operator<< from global namespace into Catch namespace
     // This is necessary because the overload of operator<< above makes
-    // lookup stop at namespace Catch
+    // lookup stop at namespace PrusaCatch
     using ::operator<<;
 
     // Use this in variadic streaming macros to allow
@@ -539,13 +540,13 @@ namespace Catch {
     ::Catch::SourceLineInfo( __FILE__, static_cast<std::size_t>( __LINE__ ) )
 
 // end catch_common.h
-namespace Catch {
+namespace PrusaCatch {
 
     struct RegistrarForTagAliases {
         RegistrarForTagAliases( char const* alias, char const* tag, SourceLineInfo const& lineInfo );
     };
 
-} // end namespace Catch
+} // end namespace PrusaCatch
 
 #define CATCH_REGISTER_TAG_ALIAS( alias, spec ) \
[... 2756 lines cut ...]
+namespace Catch = PrusaCatch;
+
+#endif  // PRUSA_CATCH
diff --git a/tests/catch_main.hpp b/tests/catch_main.hpp
index 5ab71fd..37ee2e6 100644
--- a/tests/catch_main.hpp
+++ b/tests/catch_main.hpp
@@ -6,7 +6,7 @@
 #define CATCH_CONFIG_DEFAULT_REPORTER "verboseconsole"
 #include <catch2/catch.hpp>
 
-namespace Catch {
+namespace PrusaCatch {
 struct VerboseConsoleReporter : public ConsoleReporter {
     double duration = 0.;
     using ConsoleReporter::ConsoleReporter;
@@ -49,6 +49,6 @@ struct VerboseConsoleReporter : public ConsoleReporter {
 
 CATCH_REGISTER_REPORTER( "verboseconsole", VerboseConsoleReporter )
 
-} // namespace Catch
+} // namespace PrusaCatch
 
 #endif // CATCH_MAIN
```

hyperair, 2026-09-21:
Looks like the PR fixing this issue was closed, so I'll be closing this issue as well.

closed 2026-09-21 by hyperair