prusa3d/PrusaSlicer #5401: Git Linux WX stable build failure

clytle374, 2020-12-08:
Having another build failure with WX widgets 3.0.  On Gentoo Linux with most current git source. Here is the output.
`[ 59%] Built target imgui
[ 59%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Field.cpp.o
/home/cory/sources/prusa/src/slic3r/GUI/Field.cpp: In member function ‘void Slic3r::GUI::Field::get_value_by_opt_type(wxString&, bool)’:
/home/cory/sources/prusa/src/slic3r/GUI/Field.cpp:348:38: error: ‘format_wxstr’ was not declared in this scope
  348 |                 show_error(m_parent, format_wxstr(_L("Invalid input format. It must be represented like \"%1%\""),"XxY, XxY, ..." ));
      |                                      ^~~~~~~~~~~~
`

Wish the wxwidgets would just stabilize a newer version.  Thanks
Cory

bubnikv, 2020-12-08:
I think this is now fixed on master, right?

clytle374, 2020-12-08:
Confirmed as fixed, thank you! Your efforts are appreciated.

closed 2020-12-08 by clytle374